Lora-net/packet_forwarder #7: Server Messages

MKillev, 2016-04-24:
Hi,

I have setup `Lora-net/packet_forwarder` and server;
Is there any standart way to parse the data from messages? 
Or can I send only valid json messages with the `packet_forwarder`?

On the server I receive messages like this ones:

```B5�UZ{"stat":{"time":"2016-04-24 16:32:37 GMT","rxnb":2,"rxok":2,"rxfw":2,"ackr":0.0,"dwnb":0,"txnb":0}}
+@�UZ{"rxpk":[{"tmst":2934474419,"chan":2,"rfch":1,"freq":868.500000,"stat":1,"modu":"LORA","datr":"SF7BW125","codr":"4/5","lsnr":6.8,"rssi":-67,"size":18,"data":"QBEREREAlAMEX5iCQB8ij0ZU"}]}
���UZ{"rxpk":[{"tmst":2921073603,"chan":0,"rfch":1,"freq":868.100000,"stat":1,"modu":"LORA","datr":"SF7BW125","codr":"4/5","lsnr":7.8,"rssi":-71,"size":18,"data":"QBEREREAkwMEoIEzjVYBnMfH"}]}
e��UZ{"rxpk":[{"tmst":2908126707,"chan":2,"rfch":1,"freq":868.500000,"stat":1,"modu":"LORA","datr":"SF7BW125","codr":"4/5","lsnr":6.2,"rssi":-66,"size":18,"data":"QBEREREAkgME/9mgCeWD3REa"}]}
O�UZ{"rxpk":[{"tmst":2894923683,"chan":0,"rfch":1,"freq":868.100000,"stat":1,"modu":"LORA","datr":"SF7BW125","codr":"4/5","lsnr":8.2,"rssi":-65,"size":18,"data":"QBEREREAkQMEi7mXjZsNpziU"}]}
```

Thanks in advance, any help will be appreciated.

mcoracin, 2016-04-25:
Hello,

I'm not sure I fully get your question.

The message format sent by the gateway are formatted as described in the PROTOCOL.txt file you can found in the lora-net/packet_forwarder root directory.

Typically there are few octets (4 to 12) describing the protocol version, the message type etc...
Then, after those few octets, you have the JSON string which contains the actual packet data.

SO the server has to parse the few bytes as a binary header, and then any JSON parser car handle the JSON string.

Regards

MKillev, 2016-04-25:
Hello,
Thank you for the quick response. This answers my question.

closed 2016-04-25 by MKillev

MKillev, 2016-04-26:
Hello,
One more thing. I receive the uplink messages, but it seems I do NOT
receive the PULL DOWN ack message on the server side. When I uncomment the
the line in the source of the beacon "//MSG("WARNING: [down] recv returned
%s\n", strerror(errno)); /* too verbose */" it is
shown continuously although  I am sending messages down the UDP to the
Gateway on a 5 sec basis.

When I start the util_sync I see that the packets are received but still
the beacon program is not aware of them.

Any suggestions?

Regards.