Lora-net/packet_forwarder #9: List of global_conf.json keys

Mehradzie, 2016-05-12:
Atm, there is no documentation on all the available keys on the **global_conf.json** or the functionality of each  key. 

We have multiple questions in mind that could be answered if we had a very brief documentation, since we are doing lots of guess work. There are tags that we accidentally discovered by looking at multiple different example codes and we would really appreciate it if we could have a list :smile:

Another big question in mind 
We are wondering if we could add a second **mCard** to our **Multitech Conduit** and then expand the packet forwarder to handle it by using another ` "SX1301_conf" ` key? If so, what is the limit. This would let us to listen to 16 channels simultaneously rather than 8. 

Thanks in advance for your help.

dhavalp, 2016-12-08:
I am looking for a conf file for RN2483 http://www.microchip.com/DevelopmentTools/ProductDetails.aspx?PartNO=rn-2483-pictail and suggestions for that?

cirkutninja, 2017-06-05:
@Mehradzie  Was curious to know if you had any resolution to your post? I'm also having an issue trying to figure out the functionality of the keys within the global_conf.json files and was wondering if you had an answer?

mcoracin, 2017-06-06:
There is no documentation available yet.
Until it is the case, I suggest that you ask here for keys which are not clear to you.

Almost all keys end up being a configuration of the lora_gateway HAL, which is somehow documented. You should find the answers for most of the keys.
But we can help for the others.

cirkutninja, 2017-06-07:
@mcoracin thanks for the info. I'll take a dig around in the HAL and post if I need help, as you suggest.

cdliuqiang, 2018-04-11:
@mcoracin   there is no global_conf about sx1255,   can you tell me the specific meaning of "tx_lut_x" configuration, and the difference between sx1255 and sx1257 in "tx_lut_x" configuration?

mcoracin, 2018-04-16:
@cdliuqiang sorry, I don't have those information, you should contact Semtech support on LoRa community.

closed 2021-06-24 by rbaldwin13