Lora-net/packet_forwarder #10: DownStream

MKillev, 2016-05-20:
Hello,

I am trying to issue a downlink to the forwarder but it does not work as expected. 

When I receive a packet PUSH_DATA for example, i get the tokens and try to return an acknowledge to the forwarder but on the status afterwards I get PUSH_DATA acknowledged 0.00%. This is weird because i sent it according the protocol (0x01 token token 0x01). I see that the bytes are received on the forwarder because i use the utilities to test the UDP connection and I see the 4 bytes received on the forwarder side, but are not recognized by the software. The same is valid for the PULL DATA requests and acknowledges.

Can you please assist me in some way?

Regards.

mcoracin, 2016-05-24:
Hello,

It could happen that the PUSH_ACK is getting lost if the latency between the forwarder and the server is too big.
Could you try to enable the following logs in the packet forwarder to see if the ACKs are received but ignored?
in thread_up():
...
MSG("WARNING: [up] ignored invalid non-ACL packet\n");
...
MSG("WARNING: [up] ignored out-of sync ACK packet\n");

Regards

MKillev, 2016-05-24:
Hello,
I have enabled the Message logs, but none of them appear. I also tried it with the beacon utility and the same issue is there as well. In the same time i enabled the util_sync and I see that the forwarder receives the 4 bytes i send to it. Its just it does nothing with them.

mcoracin, 2016-05-24:
Could you describe more precisely your test setup?
I guess you have a server running on the gateway, and the packet forwarder server address is set as 'localhost'?
Have you properly set the up/down ports in the JSON conf file, in line with what your server expect?

I don't understand your remark about util_sink.

Could you try, on your gateway to run both the packet forwarder and the util_ack.
Configure the packet forwarder server to localhost and up/down port to 1680.
Launch the packet forwarder, then run ./util_ack 1680

Do you see PUSH_ACK, PULL_ACK messages on the packet forwarder?

Here is what I see on my side:

--> packet forwarder logs:
JSON up: {"rxpk":[{"tmst":140991580,"chan":6,"rfch":0,"freq":867.700000,"stat":1,"modu":"LORA","datr":"SF12BW125","codr":"4/5","lsnr":10.2,"rssi":-31,"size":29,"data":"gK8BzAAAbwACSkFo27Ww3LgC1N+T5fbmCAV40/o="}]}
INFO: [up] PUSH_ACK received in 30 ms
INFO: [down] PULL_ACK received in 30 ms

--> util_ack logs:
 -> pkt in , host 127.0.0.1 (port 44547), 214 bytes, PUSH_DATA from gateway 0xB827EBFFFEA047D0
<-  pkt out, PUSH_ACK for host 127.0.0.1 (port 44547), 4 bytes sent
 -> pkt in , host 127.0.0.1 (port 44547), 215 bytes, PUSH_DATA from gateway 0xB827EBFFFEA047D0
<-  pkt out, PUSH_ACK for host 127.0.0.1 (port 44547), 4 bytes sent
 -> pkt in , host 127.0.0.1 (port 40511), 12 bytes, PULL_DATA from gateway 0xB827EBFFFEA047D0
<-  pkt out, PULL_ACK for host 127.0.0.1 (port 40511), 4 bytes sent

Is the same working for you?

MKillev, 2016-06-01:
Hello,

My test setup is as follows. I have the gateway on the one side and separately I have the server on a remote computer. I am communicating through UDP between them. I get the following log from the beacon packet forwarder:

### [UPSTREAM] ###
# RF packets received by concentrator: 0
# CRC_OK: 0.00%, CRC_FAIL: 0.00%, NO_CRC: 0.00%
# RF packets forwarded: 0 (0 bytes)
# PUSH_DATA datagrams sent: 1 (111 bytes)
# PUSH_DATA acknowledged: 0.00%
### [DOWNSTREAM] ###
# PULL_DATA sent: 0 (0.00% acknowledged)
# PULL_RESP(onse) datagrams received: 0 (0 bytes)
# RF packets sent to concentrator: 0 (0 bytes)
# TX errors: 0
### [GPS] ###
# GPS sync is disabled
##### END #####

As you can see the acknowledge is at 0% even though i am continuously sending the ack packets. And the getaway receives them, but apparently doe not recognizes them as such. And the biggest problem of mine is that the pull data is not recognized as well, which means i cannot send a message to an end device.

mcoracin, 2016-06-01:
Hello,

Could you reproduce the test I described above, which consists in running both the packet forwarder and the util_ack on the gateway itself?

closed 2016-06-06 by MKillev

mcoracin, 2016-06-06:
Hello @MKillev ,

Did you find the issue?

MKillev, 2016-06-06:
Hello,
Yes I did. I was sending the messages from the server to the gateway trough the wrong port. As soon as I figured that and sent it trough the port that the gateway was sending, everything started working!

Thank you for the support!

mcoracin, 2016-06-06:
ok, glad to read that it works now! :)