Lora-net/packet_forwarder #12: the timestamp in beacon is placed in a wrong place?!

rossihwang, 2016-05-31:
Under the beacon_pkt_fwd example

```
void thread_down(){
    ....
    beacon_pkt.payload[ 9] = 0xFF &  field_time;
    beacon_pkt.payload[10] = 0xFF & (field_time >>  8);
    beacon_pkt.payload[11] = 0xFF & (field_time >> 16);
    beacon_pkt.payload[12] = 0xFF & (field_time >> 24);
    ...
}

```
According  to the spec 1.0.1 Page 86, it should be like this. 

```
void thread_down(){
    ....
    beacon_pkt.payload[3] = 0xFF &  field_time;
    beacon_pkt.payload[4] = 0xFF & (field_time >>  8);
    beacon_pkt.payload[5] = 0xFF & (field_time >> 16);
    beacon_pkt.payload[6] = 0xFF & (field_time >> 24);
    ...
}
```

mcoracin, 2016-06-07:
Hello,

The beacon_pkt_fwd has been obsoleted with the latest release, as it was just experimental, and not tested at all.

FYI, the new lora_pkt_fwd, supports beacon, according to LoRaWAN1.1 definition.

You may need to slightly adapt it if used with LoRaWAN1.0.

Regards

closed 2016-06-07 by mcoracin

Mariiian09, 2025-04-02:
> Hello,
> 
> The beacon_pkt_fwd has been obsoleted with the latest release, as it was just experimental, and not tested at all.
> 
> FYI, the new lora_pkt_fwd, supports beacon, according to LoRaWAN1.1 definition.
> 
> You may need to slightly adapt it if used with LoRaWAN1.0.
> 
> Regards

Thanks for your response. I understand that beacon_pkt_fwd is obsolete and that lora_pkt_fwd supports beacons based on LoRaWAN 1.1. However, I am using LoRaWAN 1.0.3, and my end device is not receiving beacons.

You mentioned that some adaptations are needed for LoRaWAN 1.0. Could you clarify what specific changes are required to make lora_pkt_fwd work correctly with LoRaWAN 1.0?

Any guidance would be greatly appreciated!