Lora-net/packet_forwarder #15: ERROR

verogeorl, 2016-07-25:
Hi,
Sorry if this section is not suitable to leave messages about problems we encoutered while following your tutorial but I don't where I can leave my messages (is there a forum?).

I tried following your tutorial but I always get the same error when I want to run lora_pkt_fwd.
I always get: 
[image]
Failed to start the concentrator, any idea why I have this?

Thanks

Mehradzie, 2016-07-26:
If you're using a Kerlink GW, in the shell try 
`modem_on` 
And them press tab for autocomplete. The upcoming script if you run it, will start the modem on the Kerlink GW. 
This could be one of the issues. Let us know

verogeorl, 2016-07-26:
No actually I'm using a Raspberry 3 and trying to find where the GPIO is used in the code to see if it's compatible.

Thanks for your answer

Mehradzie, 2016-07-26:
Np. 
If you're going through the code for the packet forwarder, make sure to check the for another project called *LoRa gateway* from the same developer (Semtech). Most of the packet forwarder stuff are actually are hidden there, and packet forwarder needs the LoRa gateway code to compile. 
https://github.com/Lora-net

I haven't done anything outside of Kerlink and Multitech gateways so that's just a hint.

mcoracin, 2016-08-17:
Hello @verogeorl,

Which version of the gateway board do you have?
If you are using a v1.0 (the one without a FPGA), you need to reset the SX1301 before launching the packet forwarder.
This can be done using the script reset_lgw.sh provided with the HAL (Lora-net/lora_gateway).

Please also send the wiring you are using between the RPi and the gateway board.

Regards