Lora-net/packet_forwarder #16: duty cycle handling

queifaro, 2016-09-09:
Hi,
how the pktfwd or iC880 (design ref 1) handle the duty cycle limit on TX ? source code seams no taking into account that ....
Last LoRaWan spec confirm that at the moment only the duty cycle is consider and not LBT+AFA 
Thanks
F

mcoracin, 2016-09-16:
Hi,

It is not handled by the packet forwarder at the moment, and there is no plan to handle that in a near future.
For now, it is up to the gateway vendor/developer/provider/... to implement this part.

Regards

closed 2016-09-16 by mcoracin

Yaoshicn, 2017-07-17:
Hi, @mcoracin ,

Did you mean that the duty-cycle is set in hardware by gateway vendor/developer/provider/...?
Thanks in advance.

Best Regards

mcoracin, 2017-07-18:
Hi @Yaoshicn,
No, it's not set in hardware. It can be handle in SW by the gateway, or even on the server side.
Best regards,
Michael

Yaoshicn, 2017-07-18:
Hi, @mcoracin ,

Thank you for your reply. Ask one more question, is the regulation of LoRaWAN duty-cycle still applicable when the whole network not connected with public LoRaWAN network? (e.g., for research purpose)

Thank you for your help and patience.

Best Regards,
Yaoshicn