Lora-net/packet_forwarder #17: Class-C  ACK:TOO_EARLY

qingchuwudi, 2016-09-13:
Hi,
   I'm testing **Class-C**, and gateway return an ACK to server `TOO_EARLY`. But everything is ok, if server uses **Class-A** and mote uses **Class-C**.
   I have set `imme=true` within `txpk`. 
  For example:

  - This is all right (Class-A)

  ```JSON
  {
    txpk:{
      imme:false,
      tmst: RecvTime + RX1/RX2,
      ...
      }
  }
  ```
  - ACK : `TOO_EARLY` (Class-C)

  ```JSON
  {
    txpk:{
      imme:true,
      ...
      }
  }  
  ```

What's wrong with it? What should I do with gateway?

qingchuwudi, 2016-09-13:
```
JSON down: {"txpk":{"time":"2016-09-13T16:05:15.286674Z","size":15,"rfch":0,"modu":"LORA","ipol":"true","imme":"true","freq":433.3,"datr":"SF12BW125","data":"YAgAAAEAfAACiPydFEuT","codr":"4/5"}}
INFO: [down] a packet will be sent on timestamp value 2093252875 (calculated from UTC time)
src/jitqueue.c:314:jit_enqueue(): ERROR: Packet REJECTED, timestamp seems wrong, too much in advance (current=2094548812, packet=2093252875, type=1)
ERROR: Packet REJECTED (jit error=2)
...

JSON down: {"txpk":{"time":"2016-09-13T16:07:13.556752Z","size":15,"rfch":0,"modu":"LORA","ipol":"true","imme":"true","freq":433.3,"datr":"SF12BW125","data":"YAgAAAEAfQACjOE/P1k0","codr":"4/5"}}
INFO: [down] a packet will be sent on timestamp value 2212252905 (calculated from UTC time)
src/jitqueue.c:314:jit_enqueue(): ERROR: Packet REJECTED, timestamp seems wrong, too much in advance (current=2212818957, packet=2212252905, type=1)
ERROR: Packet REJECTED (jit error=2)
...
```

Why `packet < current` ?

closed 2016-09-13 by qingchuwudi

batulzii, 2017-05-05:
can you share how did you done?

qingchuwudi, 2017-05-08:
@batulzii Given a wrong `txpk` package with format and value error.

divum21, 2018-08-01:
@qingchuwudi 
Please tell me how you resolved that timestamp problem.!
I am facing this issue for longtime