Lora-net/packet_forwarder #29: No rule to make target

wofof032, 2017-02-24:
I try to make the paket forwarder I get

pi@WLo-LoRa-GWY:~/packet_forwarder $ sudo make all
make all -e -C lora_pkt_fwd
make[1]: Entering directory '/home/pi/packet_forwarder/lora_pkt_fwd'
make[1]: *** No rule to make target '../../lora_gateway/libloragw/inc/loragw_hal.h', needed by 'obj/lora_pkt_fwd.o'.  Stop.
make[1]: Leaving directory '/home/pi/packet_forwarder/lora_pkt_fwd'
Makefile:11: recipe for target 'all' failed
make: *** [all] Error 2
pi@WLo-LoRa-GWY:~/packet_forwarder $

Any ideas?
Thanks

johanstokking, 2017-02-24:
You need to make the `lora_gateway` first for your platform. Also, make sure that the relative path to `lora_gateway` is set correctly. If the header can be found at that location, make won't complain about a missing target.

mcoracin, 2017-03-09:
This has been answered, thanks @johanstokking , I close the issue.

closed 2017-03-09 by mcoracin