Lora-net/packet_forwarder #32: A lot of CRC_FAIL

akm-barnaul, 2017-03-09:
Hello!
I'm using:
- iC880A (USB, with additional power supply) with a Cubieboard A20 (Ubuntu server)
- lora_gateway 3.10 (I know that is not the recent version, but I have only the USB-version of iC880A)
- packet_forwarder 2.10
Gateway is 2 m far from a node (Arduino + HopeRF 95).
A lot of packets are failed to forward because of CRC_FAIL error. thethingsnetwork.org application data log skips 1 or 2, or even 3 packets (according to counter).
Also, when I use a debug build of libloragw I see a lot of failed packets (third byte is 7).

So, the question is: how can I get rid of CRC_FAILED?

johanstokking, 2017-03-09:
This is almost always an RF issue. Check antennas and wiring. Also it might be slight differences between the frequencies. Note also that SF11 and SF12 have known problems in older versions that cause CRC errors.

mcoracin, 2017-03-09:
Just in case, check that the 'lorawan_public' field of the global_conf.json file is set to true. It should be the case by default.
And of course ensure that the corresponding configuration is done on the Hope RF side (public/private).
When there is a mismatch we get only around 10% of the packets.

akm-barnaul, 2017-03-10:
> Just in case, check that the 'lorawan_public' field of the global_conf.json file is set to true. It should be the case by default.
And of course ensure that the corresponding configuration is done on the Hope RF side (public/private).

Checked the fields - they are set as you said

> When there is a mismatch we get only around 10% of the packets.

I get about 90% of the packets.

iC880A manufacturer advised me to reduce transmit power of my node (arduino + LMIC library + hoperf 95w), but playing with LMIC_setDrTxpow() had no effect.

akm-barnaul, 2017-03-10:
> This is almost always an RF issue. Check antennas and wiring. Also it might be slight differences between the frequencies. Note also that SF11 and SF12 have known problems in older versions that cause CRC errors.

Wiring is ok, I have about 10% of packets lost. 
Gateway and node use the same frequencies and SF7:
"frequency": 868.92,
  "modulation": "LORA",
  "data_rate": "SF7BW125",
  "coding_rate": "4/5",
  "gateways": [
    {
      "gtw_id": "eui-02d609fffe80c5cb",
      "timestamp": 1336239307,
      "time": "2017-03-10T02:46:05.125464Z",
      "channel": 4,
      "rssi": -42,
      "snr": 9
    }
  ],

lancezhangsf, 2017-03-21:
HI，akm-barnaul
     I have the same problem,  i set freq 470Mhz,  the gateway have a lot of crc error , the data was given up . i set sf=11,bw=125K,then the freq = 471.5, I have about 10% of packets lost.  But when the node from 470 to 489,the gw show a lot crc errors.

closed 2021-06-24 by rbaldwin13