Lora-net/packet_forwarder #42: packet forwarder stops receiving packet after some time

nkaramolegos, 2017-05-11:
Hello, i am using raspberry 3 v. B with the ic880a concentrator. I set everything as it is suggested and the gateway receives packets normally. However, as I observed after some time (maybe one day) the GW stops receiving packets, the program runs but it doesn't receive any packets. I believe that is some problem with the sx1301 module but i can not figured out. As somebody suggests I am checking for 2113929216 value of SX1301 time (PPS) and in this case I am shutting down the packet forwarder but this value never came up (unlike the error). Any ideas?

reixd, 2017-05-11:
@nkaramolegos take a look into this issue and workaround https://github.com/Lora-net/lora_gateway/issues/26#issuecomment-283089276

dmamalis, 2017-05-12:
@nkaramolegos have you tried to run the tests when your gateway has stopped receiving packets? Could you post the results of the spi stress utility?

nkaramolegos, 2017-05-12:
@reixd Thank you for your reply. I read this issue and I fixed the problem with the PLL (I did not change the SPI speed I just wait for some more time for PLL to start). However, I decreased SPI speed to 1 MHz (from 8 MHz) and the problem of losing packets after some time remained (so I set it back to 8 MHz). Let me give you some more information. I have connected pi3 with the ic880a module using cables (I will order the shied asap) and I am connecting to pi3 via ssh (without graphical interface) using WiFi. Also, I use a LoRaWAN node which sends data every 30 sec. My GW receives all the packets that my node sends with a very small (or no) loss but also receives some packets (maybe from other nodes with bad CRC - this is not a problem). However, after some hours GW stops receiving packets from my node and receives only bad CRC packets. If I restart the packet forwarder then my GW works fine again (without changing something to my node). 
@dmamalis I will try this and I will post the results. What debug message do I have to use? HAL and REG, or and SPI too?

nkaramolegos, 2017-05-15:
Hello again, I would like to ask, is there any maximum number of packets that the LoRaWAN GW can receive?

mcoracin, 2017-05-22:
Hello,

No, there is no maximum number of packets that the GW can receive.
Have you run the util_spi_stress utility long enough (few hours) to be sure that your SPI link is robust?

nkaramolegos, 2017-05-23:
Hello everyone, I believe that i found the solution to my problem. As I mentioned the raspberry device was connected via wifi. I changed to ethernet, and now is working without any problems. Thank you all for the help

mcoracin, 2017-05-23:
ok, good news, so I close the issue.

closed 2017-05-23 by mcoracin