Lora-net/packet_forwarder #46: PULL_RESP not received

LucaBertuolo, 2017-07-03:
Hi, 
i'm having a strange issue...
I've compile and run the lora-gateway and the packet-forwarder without any problem with a ic880a-spi concentrator.Then, I've set the UDP out/in port to localhost:1680 in global_conf.json, and I've write a very simple python script for open a UDP socket and receive/send message to the packet-forwarder.
Here my global_conf.json: https://pastebin.com/pt2TFdgv
Here my python script: https://pastebin.com/WQurTnmk

I'm able to send the PULL_ACK to the packet-forwarder anytime i get some data from the node, but the PULL_RESP never appear on the packet-forwarder log. I've also try removing the sleep from PULL_ACK and PULL_RESP, and also changing the json on the PULL_RESP.

I'm working on it because i need to send a confirmation ACK to the node, otherwise it will try to resend the packet...
Then, i've got to collect all the packet received in one hour, and every hour turn on the GSM/3G connection and send all the packet to a my server.

I'm missing some procedure to send/receive the PULL_RESP in the packet-forwarder?

mcoracin, 2017-07-03:
In your global_conf.json, you set "serv_port_down" to 1681. But you only handle port 1680 in your python script. You can use the same port (1680...) for both uplink an downlink.

LucaBertuolo, 2017-07-03:
Sorry, that was a previous try not corrected before post it...
Also with 1680 port for both uplink and downlink, the packet-forwarder always show the pull_ack in the log
https://pastebin.com/5b4xYbAy
https://pastebin.com/3h7CGV0X

The wrong character are, of course, the packet id...

mcoracin, 2017-07-03:
2 comments first:
- In your code, it is not a PULL_ACK, but a PUSH_ACK. Your are replaying to a PUSH_DATA (uplink).
- In the sock.sento() calls, you should replace (UDP_IP, OUT_PORT) with addr, that you received with sock.recvfrom(). So that it gets the internal port on which to reply.
```
                print PUSH_ACK
                sock.sendto(PUSH_ACK, addr)
```

Then, for the downlinks, I think the problem is that you send it on the uplink socket of the packet forwarder, which will do nothing with it.

You need to listen to PULL_DATA message, which is sent by the downlink socket of the packet forwarder, and send the PULL_RESP back on this socket.

LucaBertuolo, 2017-07-04:
Thanks you so much!
I'm doing some mistake with PUSH/PULL_DATA and ACK...
Now it wait a PUSH_DATA in the Upestream port, if the PUSH_DATA contain a node message, it save some value like channel and frequency, and then wait for a PULL_DATA on the Downstram port.
When it receive the PULL_DATA, it send a PULL_RESP in the Downstream port with the relative JSON and wait for a TX_ACK...

Maybe the code will be more explanatory...
https://pastebin.com/ZUfa3wa4
Ps. sorry for the comments in Italian and some test string...

Now i can finally see the PULL_RESP in the log of the packet-forwarder:
https://pastebin.com/fnDtM4G3
And now i can work to resolve the rejection of the JSON package...

Many tanks mcoracin :)

mcoracin, 2017-07-04:
Great! :)

The rejection seems to be due to tx_freq_min/tx_freq_max not set in your global_conf.json?

LucaBertuolo, 2017-07-04:
Yes, there was some error/missing value on global_conf.json, so I've changed it with the global_conf.json-example found in packet_forwarder/lora_pkt_fwd/cfg/ and....TADA!!!

INFO: Received pkt from mote: 26011D54 (fcnt=2)

JSON up: {"rxpk":[{"tmst":32739460,"chan":1,"rfch":1,"freq":868.300000,"stat":1,"modu":"LORA","datr":"SF9BW125","codr":"4/5","lsnr":3.0,"rssi":-101,"size":1,"data":"0w=="}],"stat":{"time":"2017-07-04 08:42:00 GMT","rxnb":4,"rxok":4,"rxfw":4,"ackr":0.0,"dwnb":4,"txnb":5}}
INFO: [down] PULL_RESP received  - token[64:80] :)

JSON down: {"txpk":{"imme":true,"freq":868.300000,"rfch":0,"powe":14,"modu":"LORA","datr":"SF9BW125","codr":"4/5","ipol":false,"size":1,"data":"01"}}
INFO: [down] a packet will be sent in "immediate" mode
INFO: tx_start_delay=1495 (1495.500000) - (1497, bw_delay=1.500000, notch_delay=0.000000)

The node still not receiving the packet, but the biggest problem seems resolved :)
Maybe i'm missing some parameter in the json, or i'm doing some error with channel and frequency...
I'm using a Libelium-Waspmote like node, so i think i can exclude a node problem...

By the way, many thanks for the help you gave me, now i will work to solve this problem :)
The problem with the PULL_RESP seems resolved, so i think will be better if i stop write here, due this is a packet-forwarder issue "ticket". But if you are interested or just want help me :P, i can continue to "ping" you ( or send you a PUSH_DATA :P )

mluis1, 2017-07-04:
Two issues on the downlink handling.

1st: The polarity must be inverted. Thus the field `"ipol":fase`, should be `"ipol":true`
2nd: The end device expects the downlink to be either 1 second or 2 seconds after the uplink. In your example you are sending the downlink immediately. Thus in order to have it working you must take the uplink `tmst` value and add 1000000 or 2000000 depending on which window you want to perform the downlink.

The txpk JSON should look like the following if answering on Rx1 window:
```json
{"txpk":{
"imme":false,
"tmst":33739460,
"freq":868.300000,
"rfch":0,
"prea":8,
"powe":14,
"modu":"LORA",
"datr":"SF9BW125",
"codr":"4/5",
"ipol":true,
"size":1,
"data":"01"
}}
```

The txpk JSON should look like the following if answering on Rx2 window with datarate SF12BW125:
```json
{"txpk":{
"imme":false,
"tmst":34739460,
"freq":869.525000,
"rfch":0,
"prea":8,
"powe":14,
"modu":"LORA",
"datr":"SF12BW125",
"codr":"4/5",
"ipol":true,
"size":1,
"data":"01"
}}
```

LucaBertuolo, 2017-07-04:
I've modify the python script for get the tmst from the PUSH_DATA packet, and then add the +1000000 and send the PULL_RESP with "ipol":true, but for the moment nothing change...

I think that now it's a node problem, because I'm also trying with the TTN-packet-forwarder and TTN website, and i can find the PULL_RESP on the log, but the node always says that no ACK has be received

```
INFO: [up] PUSH_ACK for server router.eu.thethings.network received in 74 ms
INFO: [down] for server router.eu.thethings.network PULL_ACK received in 63 ms
lgw_receive:1197: FIFO content: 1 70 1 5 1c
lgw_receive:1212: [1 17]
Note: LoRa packet
INFO: [up] PUSH_ACK for server router.eu.thethings.network received in 68 ms
INFO: [down] for server router.eu.thethings.network serv_addr[ic]PULL_RESP received :)
INFO: [down] a packet will be sent on timestamp value 96113892
59.61.99.5.ba.8f.8.f.0.99.10.10.0.8.0.0.a0.54.1d.1.26.20.e.0.64.4a.6d.e1.91.f2.6c.82.end
lgw_receive:1197: FIFO content: 1 9c 1 5 1c
lgw_receive:1212: [6 17]
Note: LoRa packet
INFO: [up] PUSH_ACK for server router.eu.thethings.network received in 69 ms
INFO: [down] for server router.eu.thethings.network serv_addr[ic]PULL_RESP received :)
INFO: [down] a packet will be sent on timestamp value 101254020
59.61.99.6.8.fd.a8.f.0.99.10.10.0.8.0.0.a0.54.1d.1.26.20.f.0.64.cd.bb.e3.a6.c.c0.81.end

##### 2017-07-04 10:27:29 GMT #####
### [UPSTREAM] ###
# RF packets received by concentrator: 2
# CRC_OK: 100.00%, CRC_FAIL: 0.00%, NO_CRC: 0.00%
# RF packets forwarded: 2 (56 bytes)
# PUSH_DATA datagrams sent: 3 (663 bytes)
# PUSH_DATA acknowledged: 100.00%
### [DOWNSTREAM] ###
# PULL_DATA sent: 1 (100.00% acknowledged)
# PULL_RESP(onse) datagrams received: 2 (357 bytes)
# RF packets sent to concentrator: 2 (32 bytes)
# TX errors: 0
### [GPS] ###
# GPS sync is disabled
##### END #####
INFO: [up] PUSH_ACK for server router.eu.thethings.network received in 64 ms
INFO: [down] for server router.eu.thethings.network PULL_ACK received in 73 ms

##### 2017-07-04 10:27:39 GMT #####
### [UPSTREAM] ###
# RF packets received by concentrator: 0
# CRC_OK: 0.00%, CRC_FAIL: 0.00%, NO_CRC: 0.00%
# RF packets forwarded: 0 (0 bytes)
# PUSH_DATA datagrams sent: 1 (163 bytes)
# PUSH_DATA acknowledged: 100.00%
### [DOWNSTREAM] ###
# PULL_DATA sent: 1 (100.00% acknowledged)
# PULL_RESP(onse) datagrams received: 0 (0 bytes)
# RF packets sent to concentrator: 0 (0 bytes)
# TX errors: 0
### [GPS] ###
# GPS sync is disabled
##### END #####
INFO: [up] PUSH_ACK for server router.eu.thethings.network received in 93 ms
lgw_receive:1197: FIFO content: 1 c8 1 5 1c
lgw_receive:1212: [4 17]
Note: LoRa packet
INFO: [up] PUSH_ACK for server router.eu.thethings.network received in 76 ms
INFO: [down] for server router.eu.thethings.network PULL_ACK received in 62 ms
INFO: [down] for server router.eu.thethings.network serv_addr[ic]PULL_RESP received :)
INFO: [down] a packet will be sent on timestamp value 115069212
59.61.99.6.db.cb.40.f.0.99.10.10.0.8.0.0.a0.54.1d.1.26.20.10.0.64.82.1e.80.4b.1c.e0.f3.end
lgw_receive:1197: FIFO content: 1 f4 1 5 1c
lgw_receive:1212: [6 17]
Note: LoRa packet
INFO: [down] for server router.eu.thethings.network serv_addr[ic]PULL_RESP received :)
INFO: [down] a packet will be sent on timestamp value 120209164
59.61.99.7.2a.39.30.f.0.99.10.10.0.8.0.0.a0.54.1d.1.26.20.11.0.64.e5.e0.11.b5.58.95.4.end

##### 2017-07-04 10:27:49 GMT #####
### [UPSTREAM] ###
# RF packets received by concentrator: 2
# CRC_OK: 100.00%, CRC_FAIL: 0.00%, NO_CRC: 0.00%
# RF packets forwarded: 2 (56 bytes)
# PUSH_DATA datagrams sent: 3 (665 bytes)
# PUSH_DATA acknowledged: 66.67%
### [DOWNSTREAM] ###
# PULL_DATA sent: 1 (100.00% acknowledged)
# PULL_RESP(onse) datagrams received: 2 (358 bytes)
# RF packets sent to concentrator: 2 (32 bytes)
# TX errors: 0
### [GPS] ###
# GPS sync is disabled
##### END #####
```

LucaBertuolo, 2017-07-04:
Ok, now it works with TTN-packet-forwarder, but i don't know why before it does not receive the ack...

I've found an "error" in my python script...
The "wait for TX_ACK" will cause a refuse json packet by the packet-forwarder because it create a big delay in the next packet...
```
JSON down: {"txpk":{"imme":false,"tmst":38481124,"freq":868.300000,"rfch":0,"prea":8,"powe":12,"modu":"LORA","datr":"SF10BW125","codr":"4/5","ipol":true,"size":1,"data":"01"}}
src/jitqueue.c:251:jit_enqueue(): ERROR: Packet REJECTED, timestamp seems wrong, too much in advance (current=43063638, packet=38481124, type=0)
ERROR: Packet REJECTED (jit error=2)
```

I'm thinking that the node does not receive the ACK due to a time problem...
Seems that the node send two packet in the time that I reply with an ACK to the first packet...
https://pastebin.com/fmchjFmB

mcoracin, 2017-09-20:
I close this issue as it seems not to be a gateway issue. Don't hesitate to reopen if needed.

closed 2017-09-20 by mcoracin

rajesh9949, 2021-01-07:
@LucaBertuolo was the issue solved

LucaBertuolo, 2021-04-09:
Never solved, project abbandoned.