Lora-net/packet_forwarder #47: TOO_EARLY when TOO_LATE

gotthardp, 2017-07-06:
Hello. I am getting
```
src/jitqueue.c:251:jit_enqueue(): ERROR: Packet REJECTED,
timestamp seems wrong, too much in advance (current=1213920578, packet=1213631788, type=0)
```
which is kinda weird because `packet->count_us - time_us` = 1213631788-1213920578 = -288790, which is definitely not `> TX_MAX_ADVANCE_DELAY`. The error should be TOO_LATE, not TOO_EARLY.

I guess this is an overflow error.

mcoracin, 2017-07-11:
Hello,

Here we handle a 32-bits rolling counter.
The rejection criteria are detailed in the jitqueue.c file:
https://github.com/Lora-net/packet_forwarder/blob/master/lora_pkt_fwd/src/jitqueue.c#L224

TOO_LATE: happens when the packet time is before current time, but there is not enough time to program the downlink in the SX1301 buffer.

TOO_EARLY: the calculation on counter is unsigned arithmetic, so in your case (packet->count_us - time_us) does not equal -288790, but rather 4294678506. Which is definitely greater than TX_MAX_ADVANCE_DELAY. :)

It may be a bit confusing though. But anyway, the result is the same, downlink is rejected for a good reason.

chrissnow, 2017-07-24:
Any reason not to just cast to int64 to fix this?

mcoracin, 2017-07-26:
No, the main reason is the lack of time for all the testing necessary then when we change something in that part of the code. :)
But I keep that in mind to come back on this.
If you want to test it, don't hesitate to share a patch and your test results. :)

closed 2021-06-24 by rbaldwin13