Lora-net/packet_forwarder #49: global_conf.json.US902: invalid config or does channel 64 use two radio signals?

brocaar, 2017-07-19:
Hi, I have a question related to the `global_conf.json.US902` configuration. From my understanding this configuration will configure the gateway for channel 0-7 (125kHz) + 64 (500kHz).

What I don't understand is how channel 64 (`chan_Lora_std` in the JSON) would fit on `radio: 0` as configured in the JSON.

Radio 0 frequencies (1.000.000Hz rx bandwidth):
Min: 902.200.000Hz, **Mid: 902.700.000Hz**, Max: 903.200.000Hz

Radio 1 frequencies (1.000.000Hz rx bandwidth):
Min: 902.900.000Hz, **Mid: 903.400.000Hz**, Max: 903.900.000Hz

Channel 64 (500000Hz bandwidth):
Min: 902.750.000, **Mid: 903.000.000**, Max: 903.250.000Hz

The max frequency of channel 64 (903.250.000Hz) is larger than the max frequency of radio 0 (903.200.000Hz).

Does this mean that the configuration is invalid, or does this mean that since there is an overlap of frequency range between radio 0 and radio 1, that the signal of both radios are being used?

brocaar, 2017-07-19:
I've found these two references in the HAL code (lora_gateway):

```c
#define LGW_RF_CHAIN_NB     2                   /* number of RF chains */
#define LGW_RF_RX_BANDWIDTH {1000000, 1000000}  /* bandwidth of the radios */
```

```c
/* Useful bandwidth of SX125x radios to consider depending on channel bandwidth */
/* Note: the below values come from lab measurements. For any question, please contact Semtech support */
#define LGW_RF_RX_BANDWIDTH_125KHZ  925000      /* for 125KHz channels */
#define LGW_RF_RX_BANDWIDTH_250KHZ  1000000     /* for 250KHz channels */
#define LGW_RF_RX_BANDWIDTH_500KHZ  1100000     /* for 500KHz channels */
```

code which validates the rx configuration:

```c
/* check if IF frequency is optimal based on channel and radio bandwidths */
    switch (conf.bandwidth) {
        case BW_250KHZ:
            rf_rx_bandwidth = LGW_RF_RX_BANDWIDTH_250KHZ; /* radio bandwidth */
            break;
        case BW_500KHZ:
            rf_rx_bandwidth = LGW_RF_RX_BANDWIDTH_500KHZ; /* radio bandwidth */
            break;
        default:
            /* For 125KHz and below */
            rf_rx_bandwidth = LGW_RF_RX_BANDWIDTH_125KHZ; /* radio bandwidth */
            break;
    }
    bw_hz = lgw_bw_getval(conf.bandwidth); /* channel bandwidth */
    if ((conf.freq_hz + ((bw_hz==-1)?LGW_REF_BW:bw_hz)/2) > ((int32_t)rf_rx_bandwidth/2)) {
        DEBUG_PRINTF("ERROR: IF FREQUENCY %d TOO HIGH\n", conf.freq_hz);
        return LGW_HAL_ERROR;
    } else if ((conf.freq_hz - ((bw_hz==-1)?LGW_REF_BW:bw_hz)/2) < -((int32_t)rf_rx_bandwidth/2)) {
        DEBUG_PRINTF("ERROR: IF FREQUENCY %d TOO LOW\n", conf.freq_hz);
        return LGW_HAL_ERROR;
    }
```

In addition to my above question:

Does this mean that when I configure channels, these are the maximum `if` config values (both positive and negative)?

* 125 kHz channels: 862500
* 250 kHz channels: 875000
* 500 kHz channels: 850000

Also am I right that in the above examples, these channels would exceed the radio by (500000 = `LGW_RF_RX_BANDWIDTH / 2`), so the other radio must be configured to cover this "out-of-range" part?

* 125 kHz channels: 925000 - 500000 = 425.000 Hz
* 250 kHz channels: 1000000 - 500000 = 500.000 Hz
* 500 kHz channels: 1100000 - 500000 = 600.000 Hz

Looking forward to your reply :-)

mcoracin, 2017-07-19:
Hi,
The configuration for the 500KHz channel is fine because we consider that the radio 0 RX bandwidth is 1.1MHz for a 500KHz channel (LGW_RF_RX_BANDWIDTH_500KHZ).
So, as the radio 0 center frequency is 902.7 MHz, it means that the HAL accepts a 500KHz channel in the range of 902.15 to 903.25. Our "channel 64" fits in it because it is centered on 903MHz (902.75 - 903.25).

The maximum IF config values would be:
500KHz: 300000  => (1100000 / 2) - (500000 / 2)
250KHz: 375000 => (1000000 / 2) - (250000 / 2)
125KHz: 400000 => (925000 / 2) - (125000 / 2)

Does it make sense? :)

brocaar, 2017-07-19:
Thanks for your clarification! Yes that makes perfectly sense. I thought the radio bandwidth was always fixed to `LGW_RF_RX_BANDWIDTH` / 1.000.000 Hz, but I understand now this is not the case and it depends on the channel bandwidth :-)

(And you're right, I forgot to divide the `LGW_RF_RX_BANDWIDTH_XXX` by `2` for my max IF config values).

closed 2017-07-19 by brocaar

oyj0594, 2017-10-16:
So, IF over 400000 in 125KHz leads to error?
why is that?

my conf file gives below error.

INFO: Lora multi-SF channel 0> radio 0, IF -600000 Hz, 125 kHz bw, SF 7 to 12
ERROR: invalid configuration for Lora multi-SF channel 0

it's because of -600000 Hz IF value?
However, Korea's channel plan has to use that IF value.. 
What should I do?

reissjason, 2017-10-16:
Looks like the KR920 global_conf configured radio_0 and radio_1 to the same frequency.
Move radio_0 down 200 kHz and radio_1 up 200 kHz. Adjust the IF values accordingly.