Lora-net/packet_forwarder #50: Reliability risk

Rustie0125, 2017-07-20:
Is there anyway to run the same packet forwarder in TCP/IP mode, UDP over large networks is not advisable

mcoracin, 2017-07-20:
You can use any type of link you want between the gateway and your network server. UDP is just given as a simple example.

Rustie0125, 2017-07-20:
Excuse my ignorance but how to i go about doing that ? looking for a UDP to
TCP streamer or what ?

brocaar, 2017-07-20:
@Rustie0125 maybe this could be of interest: https://docs.loraserver.io/lora-gateway-bridge/use/data/? This transforms the UDP protocol in JSON over MQTT (thus TCP). It enables you to also use TLS to encrypt all data between your gateway and MQTT broker.

mcoracin, 2017-07-20:
@Rustie0125 You can change the packet forwarder to use TCP instead of UDP. I'll let you search on the internet how to do that (google "linux socket TCP"). But you'll need a server on the other side which behave accordingly too.

I'm not sure which problem you're trying to solve? Are you experiencing issues with UDP already?
Is it a reliability issue or a confidentiality issue, or both? 
For sure UDP is not advised for a real large scale usage anyway...

closed 2017-09-20 by mcoracin