Lora-net/packet_forwarder #56: global conf file

oyj0594, 2017-09-20:
Hi, I want to use KR channel plan related gateway conf file like below.

https://github.com/TheThingsNetwork/gateway-conf/blob/26cfea86d967dfe6cd4b319e8318d64eb4469900/KR-global_conf.json

it doesn't work for my gateway (RAK831 + rpi3) even if i delete some codes about lbt.
in the json conf file, there are IF value from  -600000 to  600000.

is this could be problem?

mcoracin, 2017-09-20:
Could you be more precise about what doesn't work? Do you get any error message at startup?

oyj0594, 2017-09-21:
this is the log below.

> pi@raspberrypi:~/LoRa/packet_forwarder/lora_pkt_fwd $ sudo ./lora_pkt_fwd
> *** Beacon Packet Forwarder for Lora Gateway ***
> Version: 4.0.1
> *** Lora concentrator HAL library version info ***
> Version: 5.0.1;
> ***
> INFO: Little endian host
> INFO: found global configuration file global_conf.json, parsing it
> INFO: global_conf.json does contain a JSON object named SX1301_conf, parsing SX1301 parameters
> INFO: lorawan_public 1, clksrc 1
> INFO: 7 LBT channels configured
> INFO: antenna_gain 0 dBi
> INFO: Configuring TX LUT with 16 indexes
> INFO: radio 0 enabled (type SX1257), center frequency 922700000, RSSI offset -166.000000, tx enabled 1, tx_notch_freq 0
> INFO: radio 1 enabled (type SX1257), center frequency 922700000, RSSI offset -166.000000, tx enabled 0, tx_notch_freq 0
> INFO: Lora multi-SF channel 0>  radio 0, IF -600000 Hz, 125 kHz bw, SF 7 to 12
> ERROR: invalid configuration for Lora multi-SF channel 0

oyj0594, 2017-10-13:
@mcoracin

could you give an advice??

closed 2017-10-18 by oyj0594