Lora-net/packet_forwarder #58: Unable to recieve schedule downlink from Server Side

monkmahadeva, 2017-09-22:
Hi,
We have a risinghf gateway in which we are using version 2.0 ./lora_pkt_fwd. We have two end nodes , One was given by risinghf along with the gateway in which we are receiving the frame. On the Second end node However we are not receiving the packet hence even the queue is not getting empty either.

The logs of the second node where problem is occuring:
INFO: Received pkt from mote: 09876543 (fcnt=0)

JSON up: {"rxpk":[{"tmst":43022164,"chan":3,"rfch":0,"freq":867.100000,"stat":1,                                                                                        "modu":"LORA","datr":"SF12BW125","codr":"4/5","lsnr":8.2,"rssi":-77,"size":29,"d                                                                                        ata":"gENlhwmAAAACeBhm7eM1gZTNjCnGsHBgXjxO4Ps="}]}
INFO: [up] PUSH_ACK received in 8 ms
INFO: [down] PULL_RESP received  - token[0:0] :)

JSON down: {"txpk":{"imme":false,"tmst":44022164,"freq":867.1,"rfch":0,"powe":14                                                                                        ,"modu":"LORA","datr":"SF12BW125","codr":"4/5","ipol":true,"size":29,"data":"oEN                                                                                        lhwkwBgACMTAwMTEwMDAwMDAxMDAwMddsy2c="}}
INFO: [down] PULL_ACK received in 8 ms

INFO: Disabling GPS mode for concentrator's counter...
INFO: host/sx1301 time offset=(1506076573s:177915µs) - drift=-96µs
INFO: Enabling GPS mode for concentrator's counter.

##### 2017-09-22 10:37:15 GMT #####
### [UPSTREAM] ###
# RF packets received by concentrator: 3
# CRC_OK: 66.67%, CRC_FAIL: 33.33%, NO_CRC: 0.00%
# RF packets forwarded: 2 (47 bytes)
# PUSH_DATA datagrams sent: 3 (523 bytes)
# PUSH_DATA acknowledged: 100.00%
### [DOWNSTREAM] ###
# PULL_DATA sent: 3 (100.00% acknowledged)
# PULL_RESP(onse) datagrams received: 2 (386 bytes)
# RF packets sent to concentrator: 2 (58 bytes)
# TX errors: 0
# TX rejected (collision packet): 0.00% (req:2, rej:0)
# TX rejected (collision beacon): 0.00% (req:2, rej:0)
# TX rejected (too late): 0.00% (req:2, rej:0)
# TX rejected (too early): 0.00% (req:2, rej:0)
# BEACON queued: 0
# BEACON sent so far: 0
# BEACON rejected: 0
### [JIT] ###
src/jitqueue.c:511:jit_print_queue(): INFO: [jit] queue is empty
### [GPS] ###
# GPS sync is disabled
##### END #####

The logs of the working Node:
INFO: Received pkt from mote: 013E6354 (fcnt=1)

JSON up: {"rxpk":[{"tmst":137621676,"chan":2,"rfch":1,"freq":868.500000,"stat":1,"modu":"LORA","datr":"SF12BW125","codr":"4/5","lsnr":7.0,"rssi":-1,"size":15,"data":"gFRjPgGAAQAIEwWspahP"}]}
INFO: [up] PUSH_ACK received in 24 ms
INFO: [down] PULL_RESP received  - token[0:0] :)

JSON down: {"txpk":{"imme":false,"tmst":138621676,"freq":868.5,"rfch":0,"powe":14,"modu":"LORA","datr":"SF12BW125","codr":"4/5","ipol":true,"size":12,"data":"YFRjPgEgxAARzi8q"}}
INFO: [down] PULL_ACK received in 8 ms

##### 2017-09-22 10:38:45 GMT #####
### [UPSTREAM] ###
# RF packets received by concentrator: 1
# CRC_OK: 100.00%, CRC_FAIL: 0.00%, NO_CRC: 0.00%
# RF packets forwarded: 1 (15 bytes)
# PUSH_DATA datagrams sent: 2 (306 bytes)
# PUSH_DATA acknowledged: 100.00%
### [DOWNSTREAM] ###
# PULL_DATA sent: 3 (100.00% acknowledged)
# PULL_RESP(onse) datagrams received: 1 (170 bytes)
# RF packets sent to concentrator: 1 (12 bytes)
# TX errors: 0
# TX rejected (collision packet): 0.00% (req:3, rej:0)
# TX rejected (collision beacon): 0.00% (req:3, rej:0)
# TX rejected (too late): 0.00% (req:3, rej:0)
# TX rejected (too early): 0.00% (req:3, rej:0)
# BEACON queued: 0
# BEACON sent so far: 0
# BEACON rejected: 0
### [JIT] ###
src/jitqueue.c:511:jit_print_queue(): INFO: [jit] queue is empty
### [GPS] ###
# GPS sync is disabled
##### END #####

JSON up: {"stat":{"time":"2017-09-22 10:38:45 GMT","rxnb":1,"rxok":1,"rxfw":1,"ackr":100.0,"dwnb":1,"txnb":1}}
INFO: [up] PUSH_ACK received in 5 ms
INFO: [down] PULL_ACK received in 8 ms
^C
##### 2017-09-22 10:38:48 GMT #####
### [UPSTREAM] ###
# RF packets received by concentrator: 0
# CRC_OK: 0.00%, CRC_FAIL: 0.00%, NO_CRC: 0.00%
# RF packets forwarded: 0 (0 bytes)
# PUSH_DATA datagrams sent: 1 (113 bytes)
# PUSH_DATA acknowledged: 100.00%
### [DOWNSTREAM] ###
# PULL_DATA sent: 1 (100.00% acknowledged)
# PULL_RESP(onse) datagrams received: 0 (0 bytes)
# RF packets sent to concentrator: 0 (0 bytes)
# TX errors: 0
# TX rejected (collision packet): 0.00% (req:3, rej:0)
# TX rejected (collision beacon): 0.00% (req:3, rej:0)
# TX rejected (too late): 0.00% (req:3, rej:0)
# TX rejected (too early): 0.00% (req:3, rej:0)
# BEACON queued: 0
# BEACON sent so far: 0
# BEACON rejected: 0
### [JIT] ###
src/jitqueue.c:511:jit_print_queue(): INFO: [jit] queue is empty
### [GPS] ###
# GPS sync is disabled
##### END #####
 Can anyone help us explaining as to why we are not receiving the packet in 2nd node ???

mcoracin, 2017-09-25:
Is the second node configured with same syncword as the first one (public or private) ?

monkmahadeva, 2017-09-25:
yes my second node is configured with public syncword (0x34)

rohitkumar5293, 2017-09-26:
i am having the same problem as stated by @monkmahadeva. Can you please help me?

mcoracin, 2017-09-26:
@monkmahadeva I would need more information in order to help.

What do you mean by "hence even the queue is not getting empty either"? Which queue are you refering to?

Could you enable some debug logs to try to get more insight?
lora_gateway/libloragw/library.cfg => DEBUG_HAL=1
packet_forwarder/lora_pkt_fwd/inc/trace.h => DEBUG_JIT=1

That would be useful first to see if the packet is being sent from the gateway or not.

rohitkumar5293, 2017-09-27:
@mcoracin Can i do uplink and downlink without timer event? what i mean to ask timer event and radio events makes txdone and rxdone, using some delay can i call directly this function? Actually my MCU won't have RTC and timer event is sync with RTC. I disable the timer event and am able to do uplink properly but for my downlink my RX window is not getting open. Please suggest me how i can resolve this downlink problem?

mcoracin, 2017-09-27:
@rohitkumar5293 this is more a end-node issue than a gateway issue, so out of scope here. :)
But I'll try to give you some hints.
First of all, which MCU do you have? Is there any hardware timer on it?
To answer your question, you need to have a way to precisely open the window at the right moment, in order to get the preamble of the proper packet received.
So you'll need a way to wait, from the moment you got the TX_DONE to the moment you'll open the receive window. And you need to calibrate the "wait time".
For this you can first wait for less than 1 second, let's say 0.75s, and increase the preamble timeout value (symbTimeout parameter of SX1272SetRxConfig() function). It should be set to 6 or 8 by default, set to something like 50. You should receive the packet. Then you need to update the wait time as needed to be able to put the symbTimeout back to 6 or 8 while still receiving the packet.
You can't keep the chip in RX for too long otherwise you'll receive downlink not supposed to be received by this end-node.

fooshaw, 2017-10-25:
I have a similar problem,follow:

JSON up: {"rxpk":[{"tmst":474980124,"chan":6,"rfch":1,"freq":472.700000,"stat":1,"modu":"LORA","datr":"SF7BW125","codr":"4/5","lsnr":7.5,"rssi":-101,"size":23,"data":"AKcAAAAAAHy+AAQAAAAAer78+Ol9r74="}]}
INFO: [up] PUSH_ACK received in 6 ms
INFO: [down] PULL_ACK received in 1 ms
INFO: [down] PULL_RESP received  - token[0:0] :)

JSON down: {"txpk":{"imme":false,"tmst":487012636,"freq":471.9,"rfch":0,"powe":14,"modu":"LORA","datr":"SF7BW125","codr":"4/5","ipol":true,"size":17,"data":"ICAWS6Ovmd/Zz+nN+MTCt0s="}}

ERROR: Packet REJECTED, unsupported RF power for TX - 14

Can you help me?Thanks.

mcoracin, 2017-11-09:
@fooshaw the tw power required by your server seems not in line with what your gateway supports.

closed 2017-11-09 by mcoracin