Lora-net/packet_forwarder #61: LoRa packet-Forwarder doesn’t start

sajanikapukotuwa, 2017-10-13:
I have installed lora_gateway and the lora-packet-forwarder git repos from Lora-net in my raspberry pi 3B and made connections to my IC8808 board as indicated in the image below.

https://forum.loraserver.io/uploads/default/original/1X/2ab38cc876e5d76bacb85ea00645f0e5ba097a50.png

I compiled the lora_gateway and the lora-packet-forwarder repos successfully and produced all their binaries.

My SPI for the Raspberry Pi is activated via raspi-config. I have confirmed that SPI is active by running cat /boot/config.txt | grep spi. My problem is that sometimes the packet forwarder choses to not start and give the following error. This issue persists for long when it starts. I receive the following error message continuously.

```
pi@raspberrypi:~/packet_forwarder/lora_pkt_fwd $ ./lora_pkt_fwd 
*** Beacon Packet Forwarder for Lora Gateway ***
Version: 4.0.1
*** Lora concentrator HAL library version info ***
Version: 5.0.1;
***
INFO: Little endian host
INFO: found global configuration file global_conf.json, parsing it
INFO: global_conf.json does contain a JSON object named SX1301_conf, parsing SX1301 parameters
INFO: lorawan_public 1, clksrc 1
INFO: LBT is disabled
INFO: antenna_gain 0 dBi
INFO: Configuring TX LUT with 16 indexes
INFO: radio 0 enabled (type SX1257), center frequency 867500000, RSSI offset -166.000000, tx enabled 1, tx_notch_freq 129000
INFO: radio 1 enabled (type SX1257), center frequency 868500000, RSSI offset -166.000000, tx enabled 0, tx_notch_freq 0
INFO: Lora multi-SF channel 0>  radio 1, IF -400000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 1>  radio 1, IF -200000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 2>  radio 1, IF 0 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 3>  radio 0, IF -400000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 4>  radio 0, IF -200000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 5>  radio 0, IF 0 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 6>  radio 0, IF 200000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 7>  radio 0, IF 400000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora std channel> radio 1, IF -200000 Hz, 250000 Hz bw, SF 7
INFO: FSK channel> radio 1, IF 300000 Hz, 125000 Hz bw, 50000 bps datarate
INFO: global_conf.json does contain a JSON object named gateway_conf, parsing gateway parameters
INFO: gateway MAC address is configured to B827EBFFFE243B19
INFO: server hostname or IP address is configured to "127.0.0.1"
INFO: upstream port is configured to "1700"
INFO: downstream port is configured to "1700"
INFO: downstream keep-alive interval is configured to 10 seconds
INFO: statistics display interval is configured to 30 seconds
INFO: upstream PUSH_DATA time-out is configured to 100 ms
INFO: packets received with a valid CRC will be forwarded
INFO: packets received with a CRC error will NOT be forwarded
INFO: packets received with no CRC will NOT be forwarded
ERROR: [main] failed to start the concentrator
```

I have made sure that the SPI connections and all pins are properly connected with the help of a multimeter. I tried two methods, namely by connecting the IC8808 board to the raspberry pi with jumber cables and also using intermediate PCB specially printed to make the above connections stated in the image. In both cases, I have received successful results. But at times, when this unit choses not to work, it just doesnt work for hours and sometimes a day or so. This unpredictable behavior is really frustrating… But it also means that my connections must be right. And I can assure that there is no lose connection from the Rpi end or from the IC8808 end. Below is a photo of both when they are connected using the intermediate board- A method that has worked but sometimes dosnt work. I have also tested with multiple raspberry pi3’s and multiple IC8808 boards. This issue is common to all setups.

https://forum.loraserver.io/uploads/default/original/1X/9103ca7257c8f461739af53f5c8b5d4b16935835.jpg

I hope this repo is the place from which everyone gets their updated Packet forwarder code and lora_gateway software from. I am also worried that this is an outdated repo as not many people seem to have this issue.

How can I solve this issue? Any advices are really appreciated.

observation : Once I keep on re-starting the packet-forwarder (sometimes over a half a day!) , it does start. When it did, I has run for weeks without trouble!

eydonmlau, 2017-10-13:
Hi sajanikapukotuwa , long time ago , I have this issue too .
You can try this 
1.Before start pkt_fwd , you should reset the pin of sx1301 each time, Use reset_lgw.sh script can do it , only change the IO PIN in the script which your RPI used.
2.it's not safety to connect the IC880 with jumper, the SPI will not not work perfectly. Use an adapter board to connect both rpi and ic880 will be wisely.

Regards.
Eydonm

mcoracin, 2017-10-13:
As said by @eydonmlau:

1. Make sure the SX1301 is put out of reset using the reset_lgw.sh script, and connect the correct GPIO pin.
2. Make sure your SPI connection is reliable, using the util_spi_stress tool provided with lora_gateway repo (./util_spi_stress -t 4). It should run for hours/days without any error.

sajanikapukotuwa, 2017-10-13:
I confirm that this solved my issue! the issue persisted for many days and I was struggling. Thanks for help people.

alexanderplanar, 2017-10-19:
Hi, everyone!
I have the same problem, but in little bit different way. 
I connected the Radio Board to the beaglebone computer via SPI, but did not connect the reset to the board. When I run the packet forwarder after the computer starts it works fine.
But when I run the packet forwarder  and close the packet forwarder each time using Ctrl+C after some time several times to test my LoRa Node board, it stops work since 3 or 4 attempt. I get this error: failed to start concentrator. 
To provide the Radio board reset I tried to disconnect the power of concentrator (separated USB-cable) keeping SPI lines connected to the board, but it does not help anyway.
Is it required to provide the reset signal to SX1301 even in case turning the power off?
It seems the problem is not in the Radio Board, but,maybe, in the computer. Maybe, I should to reset the SPI some way?
I found that the reboot of the computer solves the problem.

closed 2017-11-09 by mcoracin