Lora-net/packet_forwarder #63: LoRaWan Channel

queifaro, 2017-10-20:
as per LoRaWan spec network server can dinamically define up to 16 channels on a single device using a mac command.
In global.json packet forwarder conf file there are only 8 chan_multiSF entry.

it is possible to update that entry up to 16 chan_multiSF using for example the second radio ?

If my understanding are coorect how will be possibel configure, on the fly, the GW with dinamic network server channel mask ?

Thanks in advance
B/R

nestorayuso, 2017-10-20:
Common gateways have only one SX1301 chip.
Each SX1301 can listen to 8 different channels simultaneously (with independence on using one or two sx125x chips). you can't listen to 16 channels with simple gateways (Kerlink IOT Station....).

If you want to use 16 channels in EU868 you need a gateway with 2 x SX1301 chips (Cisco, Kerlink iBTS...).
If you want to use 64 channels in US915 you need a gateway with 8 x SX1301 chips (Kerlink iBTS, Tektelic...).

richardtomkins, 2018-03-12:
We purchased a LoRaGo Port (Sandbox Electronics, 8 channels) and mated it to a Raspberry Pi 3 B.
We modified the TTN ic880a installation bash script to perform the install. We used this collection of Semtech lora_gateway and Semtech Packet_Forwarder to become operational.
Our lack of knowledge means that
- we are not certain the global_conf.json that originates from TTN is correct for our installation, 915 MHz, Canada.
On our two STM LoRa Discovery modules, we used the US 915 MHz Hybrid configuration when building our firmware.
We initially used the Semtech NA Network Server but have now setup up LoRa Server and are now using that.
We believe we have had ABP working, but OTAA is problematic (sometimes works??) and we are trying to figure out why.
Thus, we want to know a lot more about what all the entries in the gloabl_conf.json file mean and do, even though some of them we can surmise as to what they represent.
In addition, in LoRa Server, there is a configuration option in Channel Management, where the user can presumably define the required channels for use.
- I defined 0, 1, 2, 3, 4, 5, 6, 7 and OTAA did not produce any frame logs in the LoRa Server interface.
- I went back and added 64, thus, 0, 1, 2, 3, 4, 5, 6, 7, 64. Was this the correct thing to do? I am now seeing frame logs for our OTAA End Node.

I would appreciate guidance and thoughts about this, thank you.

closed 2021-06-24 by rbaldwin13