Lora-net/packet_forwarder #65: Autostart lora_pkt_fwd

sokolbiba, 2017-10-23:
Hi everyone,
Is there any way on how  to start the packet forwarder automatically in Rapsberry pi? I tried to edit the rc.local file but that seems that is not working.
Many thanks in advance.

eydonmlau, 2017-10-23:
Hi，
if you use debian 8 or debian 9 ,you can use systemd to start the packet forwarder automatically.
edit pkt_fwd.service in /lib/systemd/system/ ,then use command systemctl daemon-reload to reload service ,and systemctl enable pkt_fwd.service \ systemctl start pkt_fwd.service to start pkt_fwd
about the format of service ,you can google it

sokolbiba, 2017-10-23:
Hi @eydonmlau 
I am using raspberry pi 3 with RASPBIAN STRETCH LITE .I am not able to find the pwk_fwd.service in the /lib/systemd/system/ directory. Should I create a new one ?

closed 2017-10-23 by sokolbiba