Lora-net/packet_forwarder #73: Can't listen

gnob417, 2017-12-14:
Hi, packet_forwarder seems to work well about 20 minutes,
however, sometimes it stops to listen any packets.

it just sends status info. with interval 30sec even though nodes are sending packets.

what's the problem?

gnob417, 2017-12-16:
I read some issues about my problems, and i changed SPI_SPEED to 1Mhz or 1.5Mhz.
however, after some minutes, status of packet forwarder says like this.

> ### [JIT] ###
> # SX1301 time (PPS): 0

it's suspicious that SX1301 time maintains 0 after some minutes.
what can i do for this problem?

alexanderplanar, 2017-12-18:
The same problem. I will spy the topic.
The most interesting, the bug may appears after 1 minute after starting the packet forwarder, after 30 minutes, or may not appears at all for a day.
I changed the SPI speed to 800kHz and made a stress tests, but with no results: no SPI errors.
What a hardware do you use? I use Mircochip LoRa Gateway Radio Board and BeagleBone PC connected to the board.

gnob417, 2017-12-18:
I'm using RPI3 + RAK831. 
Do I have to manually restart the forwarder program periodically?
I don't think that the program is designed to work for just short term.

alexanderplanar, 2017-12-18:
I don't think so, the program must work all day long.
You use the different hardware than me, it seems that is not hardware issue.

alexanderplanar, 2017-12-18:
Moreover, my node sends packets every 2 seconds, the gateway just stops work at the random moment. It does not recevie, It does not transmit (but messages in debug mode: **INFO: Enabling TX notch filter**, this fact confirms that the application trys to send the data). No messages are reported to the log, the gateway silently dies. I tried to write logs, compare it, but I found nothing.

mcoracin, 2017-12-18:
Could you run the util_spi_stress util to check that the SPI link is robust?

You should be able to run the following command overnight, with no error:
`./util_spi_stress -t 4`

alexanderplanar, 2017-12-18:
@mcoracin, yes, I think I can. I will setup the system tonight with logging.

p.s. how to enable the debug messages? I change my inc/trace.h file:
#define **DEBUG_PKT_FWD**   1
#define DEBUG_JIT       0
#define DEBUG_JIT_ERROR 1
#define DEBUG_TIMERSYNC 1
#define DEBUG_BEACON    0
#define DEBUG_LOG       1

but I do not see any "lgw_send done" messages in syslog.
The packet forwarder is running as a system service, I run it via systemd-unit-file.

alexanderplanar, 2017-12-18:
**UPD.** I've decided to run the spi-stress test for a several hours to test it right now.
Since morning the bug already appered about 10 times, if the system tends to be buggy today, I have a good chances to catch it :-)

mcoracin, 2017-12-18:
@alexanderplanar , yes you can enable debug messages in the packet forwarder as you mentionned, and also in the HAL by modifying the library.cfg file. DEBUG_HAL, DEBUG_REG, DEBUG_SPI (very verbose)....

But I'm not sure that you'll see a lot of things... :-/ If it is a SPI issue, there could be an error on the SPI link which could accidentally write a register, putting the SX1301 in a bad state...
This is why it is important to check the SPI robustness.

Also, be sure that you never have more that one packet forwarder running at the same time, or any other application interacting with the SX1301 running in parallel of the packet forwarder.

gnob417, 2017-12-18:
I'm testing spi_stress right now.

so if i have a problem with spi link, what should i do?
connection between gateway and rpi3 is through 7 cables. 
is this method also can be a problem for bad spi link?

mcoracin, 2017-12-18:
@gnob417 , yes, the cables can definitely be an issue!
Could you post a picture of your setup?

gnob417, 2017-12-18:
[image]

mcoracin, 2017-12-18:
Ok, so supposing that the wires are not too bad, you can try to wrap the ground wire around the other wires, this can greatly improve the robustness! Like this:
[image]

mcoracin, 2017-12-18:
But first let the util_spi_stress run for a long time to see if you actually have errors.

gnob417, 2017-12-18:
i got error below.
is this resulting from bad spi slink?

> Cycle 29915 > error during the buffer comparison

> Written values:
>  0E  82  EE  21  9A  3B  07  D5  9E  AF  15  5D  30  73  8A  AB
>  3E  68  E7  86  21  2B  B1  A2  0F  E1  DA  F2  78  33  7F  86
>  B6  6D  A7  50  A8  AE  25  46  5D  3A  A3  8E  AE  2D  39  EC
>  95  20  72  B6  4C  23  59  5B  04  33  4D  7C  66  CC  02  1C
>  3A  A9  6C  E2  57  92  29  B5  CC  CC  43  7A  FA  7C  66  8F
>  9D  D8  46  E9  FB  9F  44  FF  D2  91  7B  38  5E  7D  55  98
>  26  C1  7A  7E  53  A3  33  20  70  76  9A  6A  F2  01  F9  8F
>  D9  3F  78  D5  DE  BC  D4  B0  4E  50  E9  AC  CD  3E  44  F4
>  FF  BE  72  53  62  A5  73  D2  1B  0D  3C  0D  0E  35  9D  E8
>  75  15  BD  53  D2  91  04  20  E1  ED  CC  AF  2B  10  A3  2A
>  CE  15  7D  30  BA  F0  02  D5  FE  3E  E2  0C  74  7F  F4  E9
>  95  B1  3C  67  43  40  87  24  2D  53  D3  58  63  76  83  31
>  8B  00  62  45  F1  64  1A  EF  A3  FD  FB  17  7C  F0  00  11
>  A1  3C  78  E4  7D  FF  09  AA  52  DC  03  B5  53  86  E7  DE
>  86  49  24  77  AD  3E  66  50  3B  62  67  B8  52  67  C9  F3
>  A4  42  D8  21  41  E1  CB  94  BD  CE  49  10  54  30  EF  DB
>  79  13  52  27  51  B9  77  8D  1B  DF  45  6D  46  0E  60  EA
>  50  38  0B  92  19  D7  26  D7  A5  6F  E7  FA  A0  D6  D5  19
>  E9  27  40  3B  E0  B8  C8  FB  97  0D  68  DD  1B  C9  C8  6C
>  01  D3  FE  1B  AA  24  F2  50  93  D9  4A  33  B0  1F  4D  99
>  46  8D  D4  27  45  9C  22  DC  A9  8B  BA  C5  54  82  31  55
>  55  2F  70  00  53  62  50  E6  3C  9A  1A  EC  B9  67  85  FF
>  F4  5A  26  3A  F6  49  16  A0  D4  D0  65  28  52  96  7D  A8
>  C5  EE  A8  18  50  F8  FE  8C  92  18  78  4B  7F  FE  4A  74
>  58  71  AE  4E  BA  C4  EE  8E  95  53  B6  E7  E9  33  8F  AE
>  21  37  C6  72  2F  C5  FE  C1  DD  77  0C  5D  75  57  D1  CD
>  C8  7F  1B  82  43  0A  10  D8  5D  C6  C0  47  F9  4F  F5  1B
>  87  BC  8D  B6  81  8B  78  5E  02  84  BB  77  DB  8C  44  A3
>  0B  60  25  4F  6A  35  27  C7  FB  E7  0E  F5  37  04  10  BE
>  C0  9D  74  41  28  EC  9F  2B  71  5B  A2  4C  E7  E7  F0  F3
>  47  15  42  B1  4B  69  78  46  51  87  3B  88  8B  4B  46  4B
>  E8  BA  8C  11  A7  2B  3C  18  86  DE  64  6E  C5  54  61  0C
>  6A  A3  BD  B5  0C  36  FB  5D  BD  37  E5  48  82  2B  93  6B
>  E6  1F  7C  8D  4A  B8  A5  D1  96  09  3F  5C  5E  A0  68  C8
>  43  26  7D  4F  5C  78  AD  19  AF  92  61  32  BE  F4  9D  A4
>  13  19  31  5D  D1  D6  2E  67  DF  6D  C3  3D  0D  2C  05  50
>  52  82  A0  AE  FB  4D  C7  AA  DF  28  DC  9D  1C  79  41  2F
>  92  72  8C  63  48  BB  CB  28  28  8E  65  36  BA  6B  86  0C
>  ED  26  BA  E8  73  81  93  53  A9  6F  F0  C5  E9  32  F4  7B
>  A4  81  DF  ED  3C  AA  15  64  38  7A  9A  F3  E5  21  FF  D3
>  47  BA  BB  BB  3B  4E  0E  E5  BE  FE  AA  A7  30  9F  22  D5
>  20  01  C2  5C  AB  D7  C0  E4  51  5B  D7  37  7C  D6  0A  C3
>  90  C5  7E  CC  14  8C  B1  D2  8B  5B  79  BB  FA  9B  90  1A
>  9D  52  76  48  29  37  2C  7B  92  03  B2  0E  DA  BC  D1  6A
>  81  50  36  95  DC  E7  67  67  43  E0  23  3D  7C  B3  58  19
>  06  CE  61  2F  05  8E  AA  97  91  5C  A5  6B  18  77  D6  9A
>  C7  0C  2F  A3  F4  97  0B  37  77  2E  74  F3  E1  CC  0C  E7
>  9B  6E  17  A0  FC  C1  38  8D  1E  DD  F9  36  54  CF  D0  1B
>  DB  00  BF  CF  97  CA  06  0E  F8  7B  02  D9  47  0E  C1  E2
>  7C  D8  83  78  99  BB  06  B7  98  FF  EE  ED  CE  BE  08  A9
>  BE  C7  79  55  91  7F  64  89  FA  66  63  42  74  24  24  F1
>  FC  A7  69  95  62  6F  4D  FB  6E  3B  E8  3C  F9  F0  E6  B8
>  B8  5F  0D  49  DE  71  D3  D9  D7  36  1B  4C  5A  3F  3D  56
>  E7  A6  EB  49  16  38  44  84  73  2C  C1  6D  1D  A7  25  D5
>  06  32  1E  E4  A4  F1  BD  7B  27  D8  C7  81  18  04  D7  FF
>  AB  C3  48  C1  FB  8D  45  6F  B9  06  DC  D6  AD  01  AB  B3
>  33  CA  98  D7  BB  55  53  E3  2E  1A  64  46  1F  3C  45  CA
>  FF  8D  8B  FA  1A  D0  69  D4  D7  45  AA  84  46  56  38  7A
>  20  D0  51  DB  25  A4  BE  53  BF  23  99  DE  5F  DE  A8  5E
>  6C  33  58  86  03  C2  5A  DA  07  05  5F  4E  5B  97  C8  7B
>  67  19  56  8C  BE  15  E0  7D  38  79  5B  97  58  03  F5  C4
>  36  4D  4A  39  0F  A5  14  17  AA  73  65  05  0A  2D  80  71
>  46  D6  FD  04  EB  DD  81  23  57  DC  BA  AF  DF  AF  73  15
>  FD  BD  4F  0C  62  63  23  0C  D6  88  11  E0  B5  91  51  FC
[... 107 lines cut ...]
>  A4  81  DF  ED  3C  AA  15  64  38  7A  9A  F3  E5  21  FF  D3
>  47  BA  BB  BB  3B  4E  0E  E5  BE  FE  AA  A7  30  9F  22  D5
>  20  01  C2  5C  AB  D7  C0  E4  51  5B  D7  37  7C  D6  0A  C3
>  90  C5  7E  CC  14  8C  B1  D2  8B  5B  79  BB  FA  9B  90  1A
>  9D  52  76  48  29  37  2C  7B  92  03  B2  0E  DA  BC  D1  6A
>  81  50  36  95  DC  E7  67  67  43  E0  23  3D  7C  B3  58  19
>  06  CE  61  2F  05  8E  AA  97  91  5C  A5  6B  18  77  D6  9A
>  C7  0C  2F  A3  F4  97  0B  37  77  2E  74  F3  E1  CC  0C  E7
>  9B  6E  17  A0  FC  C1  38  8D  1E  DD  F9  36  54  CF  D0  1B
>  DB  00  BF  CF  97  CA  06  0E  F8  7B  02  D9  47  0E  C1  E2
>  7C  D8  83  78  99  BB  06  B7  98  FF  EE  ED  CE  BE  08  A9
>  BE  C7  79  55  91  7F  64  89  FA  66  63  42  74  24  24  F1
>  FC  A7  69  95  62  6F  4D  FB  6E  3B  E8  3C  F9  F0  E6  B8
>  B8  5F  0D  49  DE  71  D3  D9  D7  36  1B  4C  5A  3F  3D  56
>  E7  A6  EB  49  16  38  44  84  73  2C  C1  6D  1D  A7  25  D5
>  06  32  1E  E4  A4  F1  BD  7B  27  D8  C7  81  18  04  D7  FF
>  AB  C3  48  C1  FB  8D  45  6F  B9  06  DC  D6  AD  01  AB  B3
[... cut ...] 53  E3  2E  1A  64  46  1F  3C  45  CA
>  FF  8D  8B  FA  1A  D0  69  D4  D7  45  AA  84  46  56  38  7A
>  20  D0  51  DB  25  A4  BE  53  BF  23  99  DE  5F  DE  A8  5E
>  6C  33  58  86  03  C2  5A  DA  22  D2  C0  8A  0A  82  42  9E
>  60  6D  C0  0F  5B  C9  13  47  3F  7C  C1  E9  92  D6  1A  34
>  25  E1  E5  26  E4  50  33  06  22  F3  90  2D  75  D2  CB  D5
>  3F  8B  E4  9A  55  F7  E2  94  74  A3  7D  06  79  97  3A  9E
>  79  1F  C5  5D  6F  F8  63  92  EC  F3  BF  61  C5  8A  37  04

mcoracin, 2017-12-18:
@gnob417 yes.
So now you can try to improve the wires as suggested to see if it helps.

alexanderplanar, 2017-12-18:
Yes, confirm. The same problem on 69996 cycle, after about 30 min.
Interesting, that the error affects to the whole half-buffer (high 512 bytes).
The lowest 512 bytes are written with no errors, but the next 512 bytes are completely corrupted.
[image]

Written values:
 F1  37  9C  7F  AB  D1  10  08  03  05  14  7F  65  9C  D9  78 
 3A  9D  C7  EA  27  0C  AC  27  AE  00  D6  6E  3C  AC  C9  2E 
 E4  65  AD  8F  37  BD  97  3A  C3  AB  BA  28  47  93  A0  82 
 30  68  6C  57  74  18  7E  22  18  55  90  54  01  5A  82  E5 
 BF  2F  74  F6  ED  0B  31  B0  B7  EB  D8  FE  7E  79  80  AE 
 E1  EC  05  55  04  83  77  1C  D8  07  71  DA  61  F3  BF  21 
 23  34  17  10  3F  48  C0  F6  33  98  F5  B1  11  75  5F  F2 
 62  64  47  66  E8  BE  83  C0  C6  F4  9A  27  E7  5A  48  0A 
 8E  60  1A  CD  A8  DA  C4  DC  73  B9  8D  84  2E  ED  77  90 
 51  BE  F7  39  7D  7A  FA  43  6E  94  6A  55  EE  B3  60  7C 
 13  7A  4A  BB  55  0E  97  C8  C7  25  4C  F5  12  C3  86  63 
 82  7D  9D  FF  F7  97  42  65  2B  AC  BA  1A  5F  1A  96  72 
 95  E0  2E  EA  EE  C5  B2  B5  EA  FE  AB  FC  C2  31  60  44 
 AE  FD  43  A5  94  85  0A  BF  31  C4  D9  91  DF  70  03  74 
 50  31  5E  3F  F7  10  F4  E1  0E  9F  DE  D0  D0  3E  14  7E 
 3B  57  23  CF  DC  2D  8E  0E  F2  68  9F  D1  D8  A2  45  28 
 D4  A3  67  CB  B3  5C  AC  C1  FB  8A  92  CC  C8  A6  4A  03 
 FE  6E  D2  DA  9B  61  E8  8D  C9  87  5E  A1  2A  A3  C9  FE 
 46  31  C9  F9  8D  75  BB  88  00  4D  54  C8  F3  9F  CC  F1 
 0D  9E  CC  A8  FF  B4  36  C8  3C  94  69  66  38  33  64  7E 
 64  2D  78  F1  A2  33  79  A2  80  CE  6B  73  6D  37  65  7A 
 D5  31  22  D5  E5  58  9D  21  ED  07  87  25  3A  EB  A3  9E 
 18  1B  8F  BB  4E  08  5D  CE  D6  C8  42  43  FF  A7  BD  D5 
 D8  E0  AA  BD  38  47  DF  25  4E  66  4A  88  52  EE  26  6A 
 09  B5  25  58  BE  83  26  94  4B  68  D8  4B  0F  95  20  E7 
 75  CA  A5  AE  11  84  D3  60  EA  1E  E8  3C  0C  0F  A7  15 
 C4  CC  6D  82  4F  94  17  9B  FC  EF  E6  0C  84  06  F3  FA 
 D0  98  A8  E1  1C  7B  41  07  99  2A  43  A5  39  EA  BB  FD 
 B7  28  80  06  BC  97  A1  B9  86  87  C5  0A  8D  B8  04  5D 
 51  AC  3F  6D  28  80  74  C1  AA  B8  67  E3  A2  22  E1  59 
 4A  61  60  07  F8  01  C0  7E  89  85  88  16  3D  8D  74  8E 
 39  B3  FC  61  33  70  23  DE  28  8A  C1  CB  AC  A2  24  F6 
 03  84  FD  FB  86  BD  79  0F  42  02  25  80  8F  99  0E  C8 
 4C  0A  2A  80  7B  4D  5E  A3  D7  1F  6E  83  C2  93  79  C5 
 17  77  C1  9D  34  3A  AC  77  3C  D2  F7  CB  6B  05  94  B8 
 10  BE  38  8B  0B  96  2E  E2  B5  9D  65  77  30  DE  3D  47 
 55  FE  E5  8A  38  91  01  75  63  F8  40  CF  FD  D4  87  0D 
 92  BF  98  9D  55  C7  7F  0A  64  E4  82  94  C3  BF  DB  18 
 BD  C0  A2  F5  52  A3  6A  B5  9B  AB  84  99  7F  0B  A6  12 
 CA  3F  AF  1F  06  2F  2A  6A  13  AC  FE  D6  6B  D9  EF  28 
 9A  91  1D  EC  35  88  A1  D0  33  26  69  B2  31  10  C4  FC 
 4F  74  1B  55  A3  45  BF  B6  F1  BD  8D  5C  96  7C  84  30 
 0D  A2  1C  42  2A  BE  13  5D  E4  7C  0F  15  8C  D4  11  DB 
 48  2D  30  EB  72  EF  A1  64  AC  2E  C0  43  AA  45  73  B8 
 E7  90  FA  11  4E  0D  6E  32  8A  7D  47  16  51  59  F2  99 
 86  22  84  F8  12  26  5C  BE  54  1D  01  FF  62  75  B7  49 
 05  B1  5A  53  BF  C8  85  49  45  CC  5F  97  25  51  30  AB 
 74  B5  A4  86  DB  00  44  2F  1D  46  2E  7F  BB  E5  C8  C0 
 97  22  13  56  EA  98  9F  30  64  FE  C7  8A  50  F7  35  C4 
 AC  D9  4A  87  DA  8E  B7  F7  D4  E5  77  8F  CB  3F  4F  62 
 62  62  B8  4C  FA  57  7C  5F  55  43  E9  A5  3B  1E  69  E7 
 F8  B3  6F  D2  42  26  C9  16  0B  40  A6  D6  80  F5  38  E2 
 58  F0  2E  52  47  AB  B1  9D  EE  9A  42  29  B9  AC  11  B1 
 5F  80  83  A1  A6  4C  B8  B1  8D  5E  88  0D  53  C0  EF  AB 
 B1  1D  FE  F8  C8  AF  95  B7  4A  D8  E0  03  84  F1  B4  E3 
 71  37  85  17  83  3D  C9  10  9B  51  1D  EE  11  0C  9A  C2 
 2A  98  BB  F2  47  50  A9  91  28  8A  94  AC  7B  48  90  ED 
 7F  15  04  03  52  CD  13  ED  1E  31  DB  30  3D  75  F2  67 
 0D  AD  5A  55  FE  03  E6  26  8D  7B  D3  09  C3  63  F6  43 
 78  FA  46  CA  C8  59  B7  E6  8A  92  16  C8  08  09  2F  15 
 B6  89  6A  B4  8D  51  DB  1A  CC  AE  23  8F  11  19  D2  89 
 14  18  53  DC  72  0A  C2  FC  9C  D9  C4  A4  E2  F4  BA  98 
 7D  24  4D  0A  75  28  25  41  D6  48  D1  E7  62  A3  70  76 
 BC  C3  52  2E  CD  14  2A  69  ED  EF  0E  CF  E3  C8  68  60 
[... 41 lines cut ...]
 F1  FF  0C  49  A7  0D  75  2D  E9  87  B4  FF  22  97  94  01 
 E7  17  22  D7  DC  77  6D  F0  67  BE  82  83  30  35  4B  21 
 34  57  6A  DB  64  E0  09  4E  67  BD  4D  89  54  E1  8B  3B 
 F8  AD  12  D4  24  80  C4  8B  3E  47  0E  6E  7C  59  90  B1 
 B1  FA  8C  15  DA  95  63  42  52  B1  CB  A7  92  56  E2  8B 
 04  F5  5F  28  75  24  B4  B3  6B  C2  21  E7  1C  B1  98  CD 
 AC  25  E2  86  BA  46  C8  0D  F7  94  B4  89  EA  96  14  EE 
 8B  74  17  00  98  CB  B3  03  8D  D5  EA  A9  86  83  76  32 
 A8  59  B9  62  9F  81  6F  96  15  23  1F  00  BA  34  EE  45 
 A8  05  46  40  D0  F9  43  5E  CE  2D  07  55  B0  7E  87  58 
 D7  40  BB  76  C2  2A  0C  D7  4E  2B  D7  08  5F  C6  4D  07 
 CB  93  47  9C  8D  8A  FA  5B  B8  01  B0  68  7F  38  C1  56 
 78  7C  CC  3A  A6  D8  12  F4  04  E9  FC  63  AF  4A  6B  7B 
 DD  B2  17  6A  3D  11  C6  F5  12  76  5D  92  AE  1E  E8  27 
 9A  B5  61  41  8D  73  35  91  5D  32  F5  0C  7C  60  87  59 
 12  9E  C4  4F  AF  8A  44  C2  00  A2  54  AF  C0  3C  D6  44 
[... cut ...]
 C7  B4  D1  2A  23  40  B7  0D  CB  62  F6  91  D8  4A  D5  91 
 75  C1  BF  30  91  DA  82  71  00  AA  3F  C3  84  23  53  4B 
 D7  24  76  FA  65  2D  07  30  90  FD  C1  68  48  96  F9  BD 
 57  B8  ED  E8  92  70  5A  93  1A  99  56  9E  BC  AA  E9  93 
 CE  5F  8D  33  8D  94  63  1D  91  24  85  D9  BB  7E  97  12 
 37  84  FB  C9  F4  55  5C  0E  EE  B3  AC  AA  5D  96  3D  2B 
 F5  CA  5F  82  5E  C2  9F  EF  E7  24  C9  A2  A3  60  B4  DA 
 E4  AF  A3  D9  04  00  E7  F2  B3  94  9C  10  2A  D9  3B  1F

alexanderplanar, 2017-12-18:
It is strange, I have a similar connections 50-70mm length on 800kHz frequency. I supposed that there no conditions for an error appearing. I even did a test, but apparently not so long as it required to catch an error.

gnob417, 2017-12-19:
I think the problem was from bad spi link.
I have no problem with spi link right now after applying your suggestion about wrapping wires.

thanks.

mcoracin, 2017-12-19:
Great! :)
In fact, wrapping the ground around the wires create kind of a shielded cable which helps to protect from various interference that the signal wires could get.

@alexanderplanar , is it ok for you too, can I close this issue?

alexanderplanar, 2017-12-19:
@mcoracin, I'm still testing the system, but I guess the bad spi-link is the issue :) I made a special hand-made cable with shielding wire for spi-link. The test was being started 2,5 hours ago, and there no errors yet.
Yes, you can close the issue. In case I face to problems I will re-open it.
[image]
I'm not sure if so hard isolation is required, but I decided it's better to be safe than sorry :))
Maybe, my approach will help someone. ))
Thanks!

alexanderplanar, 2017-12-20:
> Cycle 2297291 > did a 1024-byte R/W on a data buffer with no error
I made a 14-hour SPI-stress test on 800kHz: no errors found.
I suppose in this case I will never face to the issue again. Let's see.

gnob417, 2017-12-27:
Hi @mcoracin 
I found this kind of problem occur again..
If wifi AP is near the lora gateway, reliability of the gateway could be decreased??

rego21, 2017-12-27:
@alexanderplanar are you using a concentrator from the microchip development kit? I'm trying to connect a raspberry pi 2 to the Microchip concentrator but when i run the util_spi_test i cant read the values the test wrote...

alexanderplanar, 2017-12-28:
@rego21, yes, I use only the Microchip LoRa Gateway Radio Board from [DevKit DV164140-1](https://ru.mouser.com/ProductDetail/Microchip-Technology/DV164140-1/?qs=c0aJy%252b8dhdClMBvpudaFcQ==)
I did never use Raspberry Pi, and I use the BeagleBone Black board, but in general it is the same I think.
I guess, you should check the SPI-connections is right (MISO,MOSI,SCK,CS) and check the power. I connect  the Radio-board to the BeagleBone via extra USB-cable to deliver the +5V power.
Moreover, it is **very** important to connect the Reset line. The test will not work properly until the reset provided. Double check if you connect the Reset line correctly to the pin used in the libloragw module.
I use GPIO7 (kernel). You can change the GPIO used to the reset in **lora_gateway/reset_lgw.sh** (**IOT_SK_SX1301_RESET_PIN** macro). I don't know if it is possible to make packet_forwarder provide the reset signal by itself, and I have to reset the radio by using shell-file (reset_lgw.sh). Anyway you can create Unit-file to make the script to reset the radio board and the packet_forwarder to start automaticlly at system startup (I choosed this way).
Check and change (if required) the SPI-name (**SPI_DEV_PATH** macro) and SPI-speed (**SPI_SPEED** macro) in **src/loragw_spi.native.c** and rebuild the  **libloragw** module. The hard-coded "/dev/spi0" did not match to my BeagleBone, I had to change SPI-name to "/dev/spidev1.0" and the speed.

alexanderplanar, 2017-12-28:
Oh, if you change SPI_DEV_PATH or SPI_SPEED, you need to rebuild either the libloragw and packet_forwarder.

rego21, 2017-12-28:
@alexanderplanar thanks for the advice! yesterday i did all that but wasnt working... the reg file gave me wrong values and ervery other tests not worked at all. Today i did exactly the same procedure and it's working! Once again thanks @alexanderplanar !

alexanderplanar, 2017-12-29:
@rego21, Hmm... Maybe, the reason of such strange behaviour is in the physical effects: the long wires or external interference. Of course, double check the connectors. Try to enhance the shielding of SPI-lines, change SPI-speed to much lower values, e.g. 100kHz, and try to run the test for couple days.

rego21, 2018-02-07:
The packet forwarder program runs with no errors, but i cant receive any packets after 5 sec! Im using the microchip gateway and one raspberry pi 2. Can anyone help me?

rego21, 2018-02-09:
@alexanderplanar can you detail your connections between the microchip radio board and the raspberry pi? Thank you once again!

alexanderplanar, 2018-02-09:
@rego21, I just open the microchip's user manual for the gateway radioboard and make a appropriate connections.
See the file attached: 40001847A.pdf
See FIGURE 1-1, page 12: I use the connector (position #11 on the figure)
See TABLE 2-1, page 20: I connected the following signal lines: MOSI(22), MISO(24), SCK(26), CSN(28) and RESET(18).
Of course, the power lines (GND and VCC) must be the same as at the other side. For example, if you use Rasperry PI, you would better to power up the radio board via special micro-USB connector (Figure 1-1, position #1) - just connect the radio via cable to the Raspberry PI.
I would not recommend to supply the power for the radio via connector #11, you should use it only to connect signal lines.
As it is discovered, this radio board is very sensitive for external electrical discharges, e.g, statical electricity. You would better to put the whole board into the box (non-transparent for radiowaves), but provide the radio cable for external antenna. I think, the best way, to put the whole system into the box, and bring out only the 12VDC power, Ethernet and radio cable.

[40001847A.pdf](https://github.com/Lora-net/packet_forwarder/files/1710962/40001847A.pdf)

alexanderplanar, 2018-02-09:
Oh, by the way, I did not use Raspberry Pi, but I use BeagleBone Black.

rego21, 2018-02-09:
@alexanderplanar  Yeah i did the same and also noticed the board is very sensitive!! My radio board is power by the USB trough one of the usb ports from Rpi, however when i connect the ethernet cable or wi-fi dongle and start any test (util_pkt_logger or tx_test) that requires the use of the sx1257, after 3 sec the sx1257 are turned off... I believe the problem is due to the board sensitive, because i can do these tests through hdmi and a keyboard connected to the Rpi with no problems. But since pkt fowarder requires internet connections, i need to use wi-fi or ethernet...

x893, 2018-02-09:
See board consumption and compare with 100mA from USB.

alexanderplanar, 2018-02-10:
@rego21, yes, as @x893 said, it can be the power supply issue. If you have an oscilloscope then try to monitor the voltage on sx1301 on radio board. To detect if the power drops to a low voltage limit (see sx1301 datasheet)/a trigger can be used. 
When you detect, that the board does not listen, how many LEDs are works ? 
As I discovered, to determine if the board works is enought to make sure, that four leds work. In case the board is hangs the only one LED works and shows the power supply  is connected. The other LEDs blows out.