Lora-net/packet_forwarder #75: "downstream" port doesn't seem to make any sense

fifteenhex, 2017-12-24:
I'm writing an application that interacts with the packet forwarder and can't get it to accept packets on the "down" port.

I have a configuration like this:

<code>
{
        "gateway_conf": {
                "gateway_ID": "AA555A0000000000",
                "server_address": "localhost",
                "serv_port_up": 1912,
                "serv_port_down": 1913
        }
}
</code>

I would expect that the packet forwarder sends packets to 1912 on localhost (it does) and it listens for packets on 1913 on local host (it doesn't).

Netstat seems to show that the local address of the down port isn't right.

<code>
udp        0      0 127.0.0.1:44255         127.0.0.1:1912          ESTABLISHED 
udp        0      0 127.0.0.1:57646         127.0.0.1:1913          ESTABLISHED 
</code>

Am I mixing something up here? Do I have to send packets back to the address that I recv'd them from on the up port? If so what is the point of the down port?

johanstokking, 2017-12-24:
The forwarder uses `serv_port_down` to poll for downlink on the remote server. This allows you to split uplink and downlink traffic by having two UDP listeners on the server.

The forwarder does not listen itself; that would be a nightmare with firewalls and NAT.

fifteenhex, 2017-12-24:
Ok, I don't really get the utility in that. Maybe if you could send uplink and downlink traffic to different places.

Anyhow thanks for clearing that up for me.

closed 2017-12-24 by fifteenhex