Lora-net/packet_forwarder #78: JSON string length is difficult to determine

huntc, 2018-02-07:
I found the protocol difficult to implement given that associated JSON data provides no means having its length determined i.e. either through having a one-byte size field as per Pascal strings, or a zero byte terminator as per C strings. Instead, the receive buffer bytes must be read and the JSON braces counted in order to determine the end.

I believe that the assumption is that a receive buffer will contain the packet to be parsed in its entirety as per #77.

johanstokking, 2018-02-07:
Use a proper JSON deserializer, i.e. Parson for C or any builtin parser in frameworks.

huntc, 2018-02-07:
My observation is that the protocol is more difficult to deal with than it needs to be. As an enhancement suggestion, having a length byte or delimiter would make it much easier to deal with. Thanks.

johanstokking, 2018-02-08:
You are basically stating that JSON is harder to deal with than CSV. That's correct. I don't know how that relates to this protocol.

huntc, 2018-02-08:
Having a length byte as the first byte of a stream let’s you know in advance of how many other bytes are coming. When you call recv, you will not know at that point if you have read all of the bytes required in the absence of a length byte. 

Alternatively, having a delimeter allows you to call recv until that delimiter is returned. 

The current protocol declares a root level `}` as the delimeter. However, determining this is more difficult that having either a length byte or a zero byte delimeter.

closed 2021-06-24 by rbaldwin13

huntc, 2021-06-24:
What’s the rationale for closing?