Lora-net/packet_forwarder #79: Busy waiting in thread_up

xapon, 2018-02-07:
There is an inefficiency issue related to a loop in thread_up function that polls gateway for new packets, and then waits for 10ms if no new packets were found.
Is there any way to refactor this into event-based solution? Or at least make a time when packet was received more deterministic (because right now there is an inherent uncertainty of 10ms).

We are currently trying to implement time synchronization protocol for LoRaWAN, and this issue seems to be the largest source of time error in the whole system.
Of course, this delay can be reduced from 10ms down to 1ms by changing the constant, but in principal, this doesn't solve the issue with busy wait approach.

With physical LoRa, we expect time sync to be accurate in microsecond level, because of single-hop architecture - but the inefficient software layer doesn't allow that.

x893, 2018-02-08:
You need implement kernel module for interrupt processing.
See interrupt support for gpio (Wiring for RasPi or other).
I port packet forwarder to STM32 platform and now can use interrupt with SX1301/08 on RX and TX complete.

nestorayuso, 2018-02-08:
read this https://github.com/Lora-net/lora_gateway/wiki/SX1301-GPIO-output-configuration if you want to use some GPIOs as a interrupt source

xapon, 2018-02-08:
@nestorayuso  @x893 thanks, will look into that

closed 2018-02-08 by xapon