Lora-net/packet_forwarder #80: LoRaWAN Gateway as a Repeater

KhaledAljehani, 2018-02-16:
Hi,

How can I configure the gateway (packets forwarder) to work as a repeater?

Nodes transmit packets and gateway receive them, then the gateway forwards/transmits the packets to another gateway, until it reaches its destination.

The purpose is to cover an area with LoRaWAN network without Internet connectivity.

Best regards.

x893, 2018-02-16:
You can't configure current version as repeater.
You need create custon software.

KhaledAljehani, 2018-02-16:
I really appreciate your reply.

The custom software will be based on packet forwarder or LoRa gateway only?

Best regards.

x893, 2018-02-16:
You use packet forwarder codes for receive and transmite LORA packets.
But need create codes for routing packets to/from other gateway (now use UDP linux sockets).

KhaledAljehani, 2018-02-16:
Thanks for your help.

I will try to develop a code for adding repeater function.

berba9, 2018-03-09:
Hi @KhaledAljehani  

did you manage to set up the repeater functionality in gateways? 
I am wondering about a similar scenario and would be great if  you could share your experience or progress.

KhaledAljehani, 2018-03-09:
Hi @berba9 

Yes, I did a simple repeater that receives packets from nodes, then sets TX parameters and transmit the received packet to another repeater / gateway. The implementation supports only uplinks.

Also, the development was in the packet forwarder, it will be very easy if you did a good analysis.

Let me know if you have questions.

Best regards.

berba9, 2018-03-10:
Hi @KhaledAljehani thanks for your response. 
Is it possible that the gateways communicate with each other at regular intervals to synchronize their clocks (some sort of time synchronization protocol). As far as i understood from a brief look at the source code(lorapacket forwarder), the gps 1pps signal is used to synchronize the internal clock of the gateways. This internal clock is then used to provide raw timestamps ("tmst") and wall clock time ("time"). This is good but not very precise (jitters). Thats why i am wondering if some time synchronization protocol could be implemented, where gateways exchange information to synchronize clocks.
Best Regards :)

KhaledAljehani, 2018-03-13:
@berba9 
Sorry for replaying late.

To honest, I did not reach this level but I can assure that exchanging data between gateways is applicable.

gnob417, 2018-03-19:
Hi @KhaledAljehani .
I'm curious about whether repeater role for downlinks is possible or not.
it seems a gateway couldn't receive downlink from another gateway when i tested.

mcoracin, 2018-03-19:
@gnob417 be careful with the signal polarity you set when sending a LoRa packet from a gateway to another.
In LoRaWAN, there is an inverted polarity between uplinks and downlinks. So basically, when a Gateway sends a downlink to a device it configures the TX as "ipol=true", to match the device configuration for receiving downlinks.
But if you want to send a packet to another gateway, which expects a non-inverted polarity, you need to configure you TX accordingly. Please check the "ipol" parameter usage in the packet forwarder.

slarun, 2018-07-25:
Hi @KhaledAljehani, have you faced looping issue in the repeater? Meaning the packet transmitted by repeated again received by the repeater and got into loop. How you avoided it?

Thanks

rohitkumar5293, 2018-08-01:
Hi @KhaledAljehani ,

"Yes, I did a simple repeater that receives packets from nodes, then sets TX parameters and transmit the received packet to another repeater / gateway. The implementation supports only uplinks."

Will you please share us how you have implemented gateway as a repeater only for uplink? It would be great if you share us your code. I tried to set TX parameter and transmitting the packet but its remain unsuccessful.

luchostem, 2019-05-02:
Hi, i’ve been developing a solution for this, it’s a laboratory experiment for now.

i’m using a gateway made with a raspberry pi 3 and a rak831 concentrator.

The idea is to use a gateway with no internet connection to repeat the messages received in a frequency pool.

I modify the packet forwarder program running in the raspberry. This program has five threads that work in five different tasks and use a loragw library to communicate with the concentrator. I leave only one thread: the one who that receives the lora packets and modify it to resend the packet in other frecuency pool, used by the gateways.

About downlinks; it's something that i'm working on.

luchostem, 2019-05-02:
Two teories. send an ACK from the repetear and no confirmation from the gateway to the repeater, that means, from user. Or sending the ACK from the gateway hoping it arrived to the endnode working with RX windows

crabsun, 2020-03-25:
> Thanks for your help.
> 
> I will try to develop a code for adding repeater function.

Hi,
any update on LoRaWAN gateway as a repeater implementation?

nestorayuso, 2020-03-25:
I think the support for repeaters in LoRaWAN will be removed in v1.0.4

AbhishekLakhara0209, 2021-09-22:
Hi @luchostem ,

Have finished done the changes for the downlink?

Are you able  to receive the join accept from the repeater gateway?

closed 2022-09-30 by smtc-bot