Lora-net/packet_forwarder #82: packet_forwarder ignores unconfirmed UPLINK messages

Almaz1c, 2018-03-11:
Hi!
After successfully provided JOIN procedure by `packet_forwarder/lora_pkt_fwd`:
```
INFO: Disabling GPS mode for concentrator's counter...
INFO: host/sx1301 time offset=(1520789735s:345101µs) - drift=1945373645µs
INFO: Enabling GPS mode for concentrator's counter.

INFO: [down] PULL_ACK received in 1 ms

INFO: Received pkt from mote: 00000000 (fcnt=0)

JSON up: {"rxpk":[{"tmst":3749387,"chan":0,"rfch":1,"freq":868.100000,"stat":1,"modu":"LORA","datr":"SF7BW125","codr":"4/5","lsnr":9.2,"rssi":-71,"size":23,"data":"AAAAAAAAAAAAD343bzM4MTYPIK+tm+w="}]}
INFO: [up] PUSH_ACK received in 2 ms
INFO: [down] PULL_RESP received  - token[114:189] :)

JSON down: {"txpk":{"imme":false,"tmst":8749387,"freq":868.1,"rfch":0,"powe":14,"modu":"LORA","datr":"SF7BW125","codr":"4/5","ipol":true,"size":17,"data":"IJ7VlZIZU0QLOSUziI3BQ5M=","brd":0,"ant":0}}
INFO: tx_start_delay=1495 (1495.500000) - (1497, bw_delay=1.500000, notch_delay=0.000000)
INFO: [down] PULL_ACK received in 1 ms
INFO: [down] PULL_ACK received in 4 ms

##### 2018-03-11 17:36:07 GMT #####
### [UPSTREAM] ###
# RF packets received by concentrator: 29
# CRC_OK: 3.45%, CRC_FAIL: 96.55%, NO_CRC: 0.00%
# RF packets forwarded: 1 (23 bytes)
# PUSH_DATA datagrams sent: 1 (203 bytes)
# PUSH_DATA acknowledged: 100.00%
### [DOWNSTREAM] ###
# PULL_DATA sent: 3 (100.00% acknowledged)
# PULL_RESP(onse) datagrams received: 1 (191 bytes)
# RF packets sent to concentrator: 1 (17 bytes)
# TX errors: 0
# TX rejected (collision packet): 0.00% (req:1, rej:0)
# TX rejected (collision beacon): 0.00% (req:1, rej:0)
# TX rejected (too late): 0.00% (req:1, rej:0)
# TX rejected (too early): 0.00% (req:1, rej:0)
# BEACON queued: 0
# BEACON sent so far: 0
# BEACON rejected: 0
### [JIT] ###
# SX1301 time (PPS): 2409221
src/jitqueue.c:448:jit_print_queue(): INFO: [jit] queue is empty
### [GPS] ###
# GPS sync is disabled
##### END #####

JSON up: {"stat":{"time":"2018-03-11 17:36:07 GMT","rxnb":29,"rxok":1,"rxfw":1,"ackr":100.0,"dwnb":1,"txnb":1}}
INFO: [up] PUSH_ACK received in 1 ms
INFO: [down] PULL_ACK received in 1 ms
```
it ignores UPLINK unconfirmed messages from recently JOINED device at all. I don't see any JSON up messages about "rxpk" but only "stat" messages.
I am pretty sure that there are unconfirmed messages are exist - I can see them using 'lora_gateway/util_pkt_logger':
```
"AA555A0000000101","","2018-03-11 17:47:42.576Z",  72068748, 868500000,1, 2,"CRC_OK" , 29,"LORA",125000,"SF12"  ,"4/5",-68, +7.5,"404B66FC-07808E00-02BEBD9B-166698E6-E4CA2649-21050B4A-BC135B7B-CA"
"AA555A0000000101","","2018-03-11 17:47:48.521Z",  78011900, 868100000,1, 0,"CRC_OK" , 29,"LORA",125000,"SF12"  ,"4/5",-70,+10.2,"404B66FC-07808F00-029241AD-B39875CC-7C4433B6-C5B89831-A9B44852-2E"
"AA555A0000000101","","2018-03-11 17:47:54.345Z",  83836060, 867700000,0, 6,"CRC_OK" , 29,"LORA",125000,"SF12"  ,"4/5",-71,+11.2,"404B66FC-07809000-02D1D9BC-7F4A2701-040D18AF-12682BA0-07EABACC-0D"
"AA555A0000000101","","2018-03-11 17:48:04.613Z",  94105364, 868500000,1, 2,"CRC_OK" , 29,"LORA",125000,"SF12"  ,"4/5",-70, +7.8,"404B66FC-07809200-02F74286-51B95825-D3B7C08D-B1EFB3DB-F0DDC8A2-C9"
"AA555A0000000101","","2018-03-11 17:48:08.766Z",  98257484, 868100000,1, 0,"CRC_OK" , 29,"LORA",125000,"SF12"  ,"4/5",-70,+10.0,"404B66FC-07809300-026859EC-92190992-2E726F86-E6191DDA-D8493FA8-37"
"AA555A0000000101","","2018-03-11 17:48:12.792Z", 102284052, 867500000,0, 5,"CRC_OK" , 29,"LORA",125000,"SF12"  ,"4/5",-72, +6.8,"404B66FC-07809400-02B0EFEB-E0246961-1B3F9041-2EF6B913-7993D944-E4"
```
Here I turned off `LORAWAN_DUTYCYCLE_ON` for debugging purposes on my node to send RF packets every 5 seconds.
If I restart `packet_forwarder` I see only one "rxpk" but not every time I restart it.
Is there any `LORAWAN_DUTYCYCLE_ON` like config exist on `packet_forwarder` to prevent "rxpk" forwarding too often or something else? How can I turn it off for debugging purposes?

mcoracin, 2018-03-12:
Hello,

It seems that your gateway receives packets, but with CRC errors mostly:
```
# RF packets received by concentrator: 29
# CRC_OK: 3.45%, CRC_FAIL: 96.55%, NO_CRC: 0.00%
```

Do you have both the gateway and the devices configured with the same syncword (public/private)?
In the gateway, the configuration is 
```
lorawan_public:true
```
Which gateway is it?

You can also add the DEBUG_HAL flag to 1 in library.cfg file of lora_gateway project, to have more info on what is going-on.

mcoracin, 2018-03-12:
The similar config in LoRaMac-node is LORAWAN_PUBLIC_NETWORK, in the Commissioning.h file.
It has to be aligned with the gateway 'lorawan_public' setting.

Almaz1c, 2018-03-18:
Both gateway and node configured as public.

Almaz1c, 2018-03-26:
When I replaced SX1308 based LoRaGo Port dev board to SX1301 based iC880A all UnconfirmedMessageACKs started to be forwarded successfully.

closed 2021-06-24 by rbaldwin13