Lora-net/packet_forwarder #87: How to log info

richardtomkins, 2018-05-03:
I have looked and looked and am not able to find out how to do the following.

Many people include a log file of dialog that appears to come from the Packet Forwarder. I have seen this here and at TTN and other sites.
The first few characters of the file are INFO:
as seen below.
How are people producing this output and how is it being captured.
I realize this is probably simple, but for the devil, I cannot figure this out.

INFO: Little endian host
INFO: found global configuration file global_conf.json, parsing it
INFO: global_conf.json does contain a JSON object named SX1301_conf, parsing SX1301 parameters
INFO: lorawan_public 1, clksrc 1
INFO: Configuring TX LUT with 16 indexes
...
INFO: Contact email configured to "contact@ch2i.eu"
INFO: Description configured to "ch2i-gw02-ic880a"
INFO: Su

mcoracin, 2018-06-11:
Those are the logs generated by the packet forwarder on the console.
How to get those, depends on how you run the packet forwarder. If you run it manually, you'll see them immediatly.
Then depending on the gateway you are using and how the packet forwarder is configured to be launched, it may change.
Maybe you can give more detail on what you are using and how...?

richardtomkins, 2018-06-11:
I will respond in a couple of weeks.
My partner dropped out of the business.
I am in the process of getting set up, again.
As I consider my revised setup, I intend to do a fresh install on the Gateway.
My Gateway is a LoRaGo PORT on a RaspBerry Pi 3 B.
I will set up the Packet Forwarder (Semtech) startup as a service.

closed 2021-06-24 by rbaldwin13