Lora-net/packet_forwarder #89: Beacon being broadcast with reference GPS coordinates

hosse005, 2018-05-30:
Is there a reason the beacon is being configured with reference GPS coordinates specified in the configuration file instead of the acquired GPS coordinates?

nestorayuso, 2018-05-30:
GPS coordinates in .json file are fake/static coordinates for gateways without GPS.
 If your gateway has GPS, delete these lines from .json file

hosse005, 2018-05-30:
The code in lora_pkt_fwd.c file may need to be changed to use meas_gps_coord instead of reference_coord, for instance:

` field_latitude = (int32_t)((reference_coord.lat / 90.0) * (double)(1<<23)); `

mcoracin, 2018-06-11:
The initial reason to use the reference coordinate was to handle the case where we have multiple sectorial antennas on a same roof, to precisely differentiate the coordinates of each antenna/gateway.

But it would make sense to use the reference coordinate if it is given in the .json, and use the acquired coordinates otherwise.

closed 2021-06-24 by rbaldwin13