Lora-net/packet_forwarder #91: TX CONFIRMED

x893, 2018-06-18:
Hi,
i set LORAWAN_CONFIRMED_MSG_ON to true (on node) but not see any confirmed message from packet forwarder.
Can you epxlain how to use confirmed mode for node/forwarder ?

Thanks in advance

reissjason, 2018-06-18:
The network server will confirm the packet in a response downlink. 
The packet forwarder does not have the Session Keys to authenticate uplink or author downlink packets.

mluis1, 2018-07-03:
LORAWAN_CONFIRMED_MSG_ON true does work correctly when connected to a network server. There is no need to add comments such //Not implemented and never set to true

If you just have a packet forwarder (not connected to a network server) and do not reply to the Confirmed message setting the ACK bit to 1 then the node will try to re-transmit it again the number of specified trials (Max 8 times).

x893, 2018-07-03:
I see packet-forwarder source and can't find any code for confirm message.
Also I can't find any code in LoRaMac-node for confirmation replay.

reissjason, 2018-07-03:
Search for AckTimeout in src/mac/LoRaMac.c

The packet confirmation is handled by the network server and end-devices. The method is provided by the LoRaWAN protocol.
You should not expect to find logic for confirmed messages in the packet forwarder.
The packet forwarder is used to communicate with end-devices on the PHY layer.

mluis1, 2018-07-03:
The packet forwarder only forwards what it receives from the RF side to the network server side and from the network server side to the RF side. It doesn't manage acknowledgements nor LoRaWAN protocol.

The packet forwarder doesn't know anything about LoRaWAN. It is just a gateway/bridge.

On the LoRaMac-node project could you please take a closer look at the source code and you will find the Confirmed messages handling at several places. In plus in order to get the LoRaWAN certification the Confirmed messages must work.

Examples extracted from develop branch.

LoRaMac.c - `void LoRaMacProcess( void )` - Line 1453
```c
                else if( MacCtx.McpsConfirm.McpsRequest == MCPS_CONFIRMED )
                {
                    if( MacCtx.NvmCtx->AckTimeoutRetry == true )
                    {
                        stopRetransmission = CheckRetransConfirmedUplink( );

                        if( MacCtx.NvmCtx->Version.Fields.Minor == 0 )
                        {
                            if( stopRetransmission == false )
                            {
                                AckTimeoutRetriesProcess( );
                            }
                            else
                            {
                                AckTimeoutRetriesFinalize( );
                            }
                        }
                    }
                    else
                    {
                        waitForRetransmission = true;
                    }
                }
```
LoRaMac.c - `LoRaMacStatus_t LoRaMacMcpsRequest( McpsReq_t* mcpsRequest )` - Line 4262
```c
        case MCPS_CONFIRMED:
        {
            readyToSend = true;
            MacCtx.NvmCtx->AckTimeoutRetries = mcpsRequest->Req.Confirmed.NbTrials;

            macHdr.Bits.MType = FRAME_TYPE_DATA_CONFIRMED_UP;
            fPort = mcpsRequest->Req.Confirmed.fPort;
            fBuffer = mcpsRequest->Req.Confirmed.fBuffer;
            fBufferSize = mcpsRequest->Req.Confirmed.fBufferSize;
            datarate = mcpsRequest->Req.Confirmed.Datarate;
            break;
        }
```

wong-hao, 2020-10-15:
> Search for AckTimeout in src/mac/LoRaMac.c
>
> The packet confirmation is handled by the network server and end-devices. The method is provided by the LoRaWAN protocol.
> You should not expect to find [...]

[image] 

> PUSH_DATA acknowledged: **0.00%**
   ...
   PULL_DATA sent: 3 (**0.00%** acknowledged)

Does this mean that if I'm not connected to any network server, ACK = 0% is normal ?

closed 2022-09-30 by smtc-bot