Lora-net/packet_forwarder #92: Send packet to Class A LoRa Node without .json

lcer93, 2018-06-19:
Hi, I would like to:
1. Receive a packet from a node
2. Reply without building a json file and without passing through a server.
(I know that it is not the way LoRaWAN works, but it is simply a test.)
My problem is that I receive the packet in thread_up and I "artificially" fill a lgw_pkt_tx_s packet with the info I received from the node (from lgw_pkt_rx_s packet). Then I enqueue the new packet in thread_down and send it in thread_jit, but my node doesn't see the reply.
I write here how I fill the lgw_pkt_tx_s:
                                txpktMio.freq_hz = p->freq_hz;
				txpktMio.tx_mode = TIMESTAMPED;
				txpktMio.count_us = p->count_us;
				txpktMio.rf_chain = 0;
				txpktMio.rf_power = 14;
				txpktMio.modulation = p->modulation;
				txpktMio.bandwidth = p->bandwidth;
				txpktMio.datarate = p->datarate;
				txpktMio.coderate = p->coderate;
				txpktMio.invert_pol = true;
				txpktMio.preamble = 0;
				txpktMio.no_crc = 1;
				txpktMio.no_header = 1;
				txpktMio.size = 6;
				txpktMio.payload[0] = 0x00;
				txpktMio.payload[1] = 0x01;
				txpktMio.payload[2] = 0x02;
				txpktMio.payload[3] = 0x03;
				txpktMio.payload[4] = 0x04;
				txpktMio.payload[5] = 0x05;
Where is the mistake/s?
Thanks in advace.

reissjason, 2018-06-19:
If you schedule the packet using the received packets count_us it will have been already passed.
You will need to increase the count_us according the the expected receive window. i.e. add 1000000 to transmit in the first receive window of 1s after end of TX.

lcer93, 2018-06-20:
I tried to increase the count_us but nothing changes. Is it possible that my end-node is waiting for an ACK and in doesn't listen to my packet?

reissjason, 2018-06-20:
Try setting preamble to 8, the preamble is needed for the device to lock onto the signal.
and no_header to 0, if a header is not provided the device would need to know the size of the packet.

lcer93, 2018-06-20:
Thank you very much, I'm going to study your documentation.

mcoracin, 2018-11-16:
Issue has been answered.

closed 2018-11-16 by mcoracin