Lora-net/packet_forwarder #96: Downlink Not Working

kpkrishnaprakash, 2018-08-07:
I am using Sx 1301 with Rpi as gateway and nodemcu with Sx 1272 as node. Packet forwarder is runnnig in Rpi and example program (ttn abp) from arduino lmic master is running in node mcu.
Gateway is able to receive and forward the uplink packet from the node. I am trying to send a downlink packet from gateway to node using util_tx_test by running it simultaneously with pkt forwarder. In the gate way side  the util_tx_test is responding to the uplink packet and is successfully sending back the downlink packet. But it is not receiving in the node side.

JSON down: {"txpk":{"imme":true,"freq":866.000000,"rfch":0,"powe":14,"modu":"LORA","datr":"SF10BW125","codr":"4/6","ipol":true,"prea":8,"size":9,"data":"AAAAAABQRVLn"}}
INFO: [down] a packet will be sent in "immediate" mode
INFO: tx_start_delay=1495 (1495.500000) - (1497, bw_delay=1.500000, notch_delay=0.000000)

mcoracin, 2018-08-16:
If you node is configured to send Class-A uplinks, you need to handle the time synchronization for the downlink to be received in the RX1 window of the RX2 window.
util_tx_test does not handle that, it just sends "immediate" downlink. It also does not handle payload encryption etc...
You need a Network Server (even a basic one) to do this.

kpkrishnaprakash, 2018-08-17:
Thank you for the reply. I tried using a basic network server with encryption and 1 sec delay in downlink. Now its working.

mcoracin, 2018-08-17:
Great :)

closed 2018-08-17 by mcoracin