Lora-net/packet_forwarder #97: Typo in PROTOCOL.txt

gotthardp, 2018-08-07:
Hello. The document says
```
 imme | bool   | Send packet immediately (will ignore tmst & time)
 tmst | number | Send packet on a certain timestamp value (will ignore time)
 tmms | number | Send packet at a certain GPS time (GPS synchronization required)
```
The `imme` and `tmst` description refer to the field `time` that however seems to be replaced by `tmms`. Should `time` be replaced by `tmms` or the field `time` is to be kept with backward compatibility with the previous protocol versions?

mcoracin, 2018-08-16:
Hello,
"time" is the UTC time of the packet, only used in "rxpk", for information. It is not used for any synchronization.

"tmms" is the GPS time of the packet, to be used for Class-B downlink synchronization.

So in PROTOCOL.txt, "tmms" and "time" are described in the "rxpk" format, and it is valid. But in "txpk" only "tmms" is described, as "time" is not valid for downlinks.

gotthardp, 2018-08-16:
I agree the "time" is not valid for downlinks. The problem I reported is that the "time" is mentioned in the "txpk" description, which as you confirmed is wrong.

See line 353 and 354: they both say (in a "txpk" description) that "time will be ignored". But there is no "time" in "rxpk"
https://github.com/Lora-net/packet_forwarder/blob/master/PROTOCOL.TXT#L353

mcoracin, 2018-08-16:
ah yes, sorry. :)
That's correct, this will be fixed with next release.
Thank you for reporting this!

closed 2018-08-16 by mcoracin