Lora-net/packet_forwarder #108: AU915 beacon format

hosse005, 2018-10-29:
I think the logic in lora_pkt_fwd.c:1980 for determining the beacon RFU field sizes may not work for the AU915 region.  Regional params specs RFU1 = 3 bytes and RFU2 = 1 byte, but this logic will configure 5 and 3?

mcoracin, 2018-11-16:
The datarate of beacon for AU915 is SF10, so the logic is correct.

hosse005, 2018-11-16:
Looks like the LoRaWAN spec has changed a bit for AU915.  In the v1.0.2 regional parameters doc, it specs SF10 as you say.  However in v1.0.3a it specs SF12, then in v1.1a back to SF10, and in latest v1.1b spec it is SF12 again.

mcoracin, 2018-11-16:
ah ok. :)
Anyway, the RFU size is directly linked to the datarate, so it should be consistent

reissjason, 2018-11-16:
Unfortunately the specification did not change the RFU sizes for AU915 through any of the referenced versions.

mcoracin, 2018-11-16:
Ah, so I need to check if it is a mistake in the specifications or in the code. :)

PellegrinoA, 2019-04-16:
@hosse005 you are a genius!

You are right. I modifed  lora_pkt_fwd.c:1980 with  RFU1 = 3 bytes and RFU2 = 1 byte. And then the gateway send beacon (size 19 byte) as in the specification for AU915.

Thank you so much!!! :D

closed 2021-07-08 by rbaldwin13