Lora-net/packet_forwarder #110: GatewayID as only numbers

ToniStonde, 2018-12-04:
I noticed a little problem if using only numbers gateway_ID. Packet_forwarder seems to parse it as an int. Example from /etc/global_conf.json after parsing configs:

"gateway_ID":"333833304b005200"

"gateway_ID":3.3383330410052e+15

First is parsed correctly because there is atleast one letter, second has only numbers.

This causes that some LoRaWAN implementations interpret this as '0000000000000000'

reissjason, 2018-12-04:
The "gateway_ID" field is a hexadecimal string of 8 bytes with no delimiters allowed.
The value is only parsed if it is a string, otherwise NULL is returned and the default value '0000000000000000' is used.

`str = json_object_get_string(conf_obj, "gateway_ID");`

ToniStonde, 2018-12-04:
yeah, but my point is this. I'll explain (this happens when the global_conf.json is created from lora-global):

option gateway_ID '333833304b005200' => "gateway_ID":"333833304b005200"

option gateway_ID '3338333041005200' => "gateway_ID":3.3383330410052e+15

So if I insert gateway_ID like the upper one -> parsed OK

But if the gateway_ID is like the bottom one (for example) so it has only numbers -> it parses it like that.

Should it also accept only hex strings which do not have A, B, C, D, E, or F in the ID ?

reissjason, 2018-12-04:
Only hex strings are accepted. If a tool is creating the global_conf.json or local_conf.json it must output the gateway_ID as a string. Numeric values are not going to be parsed.

closed 2021-06-24 by rbaldwin13