Lora-net/packet_forwarder #111: How to change the CH0-CH7 upstream frequency?

zhaozhenhong5, 2019-01-19:
I am using CN470 gateway-conf. I want to change the frequency of each upstream channel. See the config file below: For example,I changed "487.1 MHz" to "472.1 MHz" directly,but it can't work.
Do someone know how to change?
##########
		"chan_multiSF_0": {
			"desc": "Lora MAC, 125kHz, all SF, 487.1 MHz",
			"enable": true,
			"radio": 1,
			"if": -300000
		},
		"chan_multiSF_1": {
			"desc": "Lora MAC, 125kHz, all SF, 487.3 MHz",
			"enable": true,
			"radio": 1,
			"if": -100000
		},
		"chan_multiSF_2": {
			"desc": "Lora MAC, 125kHz, all SF, 487.5 MHz",
			"enable": true,
			"radio": 1,
			"if": 100000
		},
		"chan_multiSF_3": {
			"desc": "Lora MAC, 125kHz, all SF, 487.7 MHz",
			"enable": true,
			"radio": 1,
			"if": 300000
		},
		"chan_multiSF_4": {
			"desc": "Lora MAC, 125kHz, all SF, 486.3 MHz",
			"enable": true,
			"radio": 0,
			"if": -300000
		},
		"chan_multiSF_5": {
			"desc": "Lora MAC, 125kHz, all SF, 486.5 MHz",
			"enable": true,
			"radio": 0,
			"if": -100000
		},
		"chan_multiSF_6": {
			"desc": "Lora MAC, 125kHz, all SF, 486.7 MHz",
			"enable": true,
			"radio": 0,
			"if": 100000
		},
		"chan_multiSF_7": {
			"desc": "Lora MAC, 125kHz, all SF, 486.9 MHz",
			"enable": true,
			"radio": 0,
			"if": 300000
		},

johanstokking, 2019-01-19:
Adjust the `if` considering the frequency of the concerning radio.

zhaozhenhong5, 2019-01-21:
Thank you. And do you know how to adjust?

cstratton, 2019-07-17:
You have to work out the offset (within the allowed range) of each IF from the corresponding radio's center frequency.

Someone has certainly already figured this out for your bandplan

closed 2021-07-08 by rbaldwin13