Lora-net/packet_forwarder #113: Not possible to schedule upstream packet using gps time

vagola, 2019-02-27:
I am getting always the same error `TOO_EARLY`. It's not metter what gps time provided. Internal current time are always a bit forward than packet time.

vagola, 2019-02-28:
@mcoracin

vagola, 2019-03-05:
[image]

reissjason, 2019-03-05:
4192340591 - 4191569750 = 770841 us
Your packet is 0.770841s too late. 
Send it early enough to the packet forwarder to be transmitted.

reissjason, 2019-03-05:
(your tmms) - (now) = time in future
9223372036854775 - 1235838755 = 9.223370801×10¹⁵ seconds in the future.
It is not supported to schedule a packet 292471169 years into the future.

vagola, 2019-03-05:
@reissjason The internal time diffrence are very slow. Even with number overflow it's not a random luck. The gap are same for any `tmms`. We have tryed to send gps time with leap seconds, w/o leap seconds, with 1-60seconds future from current gps time. Gap always the same.

reissjason, 2019-03-05:
What is the tmms vs tmst of received packets?
How does it compare to the scheduled packets?

vagola, 2019-03-06:
Here an example where i try to send `tmms` w/o leap seconds:

```
Mar  6 07:49:31 klk-lpbs-062B9E local1.notice spf2: [1551858571154254] JSON down: {"txpk":{"imme":false,"tmms":1235893769179,"freq":868.3,"rfch":0,"powe":20,"modu":"LORA","datr":"SF7BW125","codr":"4/5","ipol":false,"prea":8,"size":11,"data":"","ncrc":false}}
Mar  6 07:49:31 klk-lpbs-062B9E local1.notice spf2: INFO: [down] a packet will be sent on timestamp value 1012433679 (calculated from GPS time)
Mar  6 07:49:31 klk-lpbs-062B9E local1.notice spf2: ERROR: Packet REJECTED, timestamp seems wrong, too much in advance (current=1012588153, packet=1012433679, type=1)
```

reissjason, 2019-03-06:
This packet is late, it needed to arrive at the packet forwarder at least 155ms earlier.
1012588153 - 1012433679 = 154474 us

vagola, 2019-03-06:
It's scheduled in future for 2 seconds

reissjason, 2019-03-07:
(current_time - 2s) is in the future?

vagola, 2019-03-07:
oops, should be `+`. Anyway, it's not correctly converted to internal time. the gap must be bigger

reissjason, 2019-03-07:
We have not had issues using this feature.

From your example.
1235893769179 -> UTC | Mar 06, 2019 | 07:49:11 | UTC
Logged at Mar  6 07:49:31 at the packet forwarder. 

Timestamp is 20 seconds late?
Have you checked the clock sync on both side?

closed 2021-07-08 by rbaldwin13