Lora-net/packet_forwarder #120: A downlink was scheduled, overwritting it

vinsanitycarter, 2019-07-19:
I'm using RHF0M031 LoRa Gateway module with my custom gateway board(based Artik710),
 not the Raspberry pi platform.

I'm sending confirmed uplinks. 

I have to receive downlink.

[image]

There is no problem with unconfirmed message tranmisssion.

When i send a confirmed uplink, i can't receive downlink with warning.

A downlink was scheduled, overwritting it.
lgw_states returned TX_SCHEDULED

[image]

But, There is no ploblem with raspberry platform.
Confirmed message is transmitted to the network server well.

This problem seems to be caused by the differnce between custom board and raspberry pi.

I think that spi and reset pins are connected properly.

I think the picture below demonstrates tha the pins are properly connected.

[image]

[image]

Is there anything else to consider besides pin connections?

lichuanhe, 2020-06-01:
i have th same question ,some times the gateway can tx json down msg ,but some it can not ,the the gateway will display "WARNING: a downlink was already scheduled, overwritting it..."!  have u resove this problem ?and how 2 resolve it? ths!!

queifaro, 2021-01-22:
Hello,
we have the same situation on a PKTFWD 4.0.1/HAL 5.0.1. Seems that when TX_SCHEDULED happen the LoRaWan device ( a symply ModBus RTU sensor that send only confirmed uplink every 15 minutes) sometimes receive the downlink  ack and sometime no (in some case the device retransmit confirmed uplink until the max retransmitting number [7]).
Our LoRa GW is Linux Embedded ARM AT91SAM9260/Kernel 2.6.30 and ntpclient is configured to sync with ntpserver every 5 minutes, no GPS connected.
Has anyone identified the root cause of the problem and possibly the solution ?

Wath do you think about  of  this https://github.com/Lora-net/packet_forwarder/issues/31#issuecomment-756980474 patch proposal ? 
Thanks in advance 
B/R
Fabio

pshingala145, 2021-07-12:
Hello,

We are using Sentrius RG1xx LoRaWAN gateway with RN2903A(US) LoRaWAN modules mounted on IMP006 Board and Gotthardp LoRaWAN Network server.

Moreover, we send Confirm up-link messages to LoRaWAN network server(NS), in that up-link is being received by NS and NS also send the ACK to a gateway for the same Up-link messages, though node doesn't receive ACK from Gateway.

Could you please help us with how node and gateway synchronized for Down-link(ACK) messages?

Note: we can see the Downlink frame in the Packet forwarder logs with the Correct timestamp but node always through mac_error, not even a single time get a response from the gateway.

Receiving window Parameters are in concordance with the server configuration.

Any leads would be appreciated!

INFO: [down] PULL_RESP received - token[230:33] :)
RG1xx29F063 lora user.notice Jul 9 09:00:09 # BEACON sent so far: 0
RG1xx29F063 lora user.notice Jul 9 09:00:09 # BEACON queued: 0
RG1xx29F063 lora user.notice Jul 9 09:00:09 # TX rejected (too early): 0.00% (req:2, rej:0)
RG1xx29F063 lora user.notice Jul 9 09:00:09 # TX rejected (too late): 0.00% (req:2, rej:0)
RG1xx29F063 lora user.notice Jul 9 09:00:09 # TX rejected (collision beacon): 0.00% (req:2, rej:0)
RG1xx29F063 lora user.notice Jul 9 09:00:09 # TX rejected (collision packet): 0.00% (req:2, rej:0)
RG1xx29F063 lora user.notice Jul 9 09:00:09 # TX errors: 0
RG1xx29F063 lora user.notice Jul 9 09:00:09 # RF packets sent to concentrator: 1 (13 bytes)
RG1xx29F063 lora user.notice Jul 9 09:00:09 # PULL_RESP(onse) datagrams received: 1 (172 bytes)
RG1xx29F063 lora user.notice Jul 9 09:00:09 # PULL_DATA sent: 3 (100.00% acknowledged)
RG1xx29F063 lora user.notice Jul 9 09:00:09 ### [DOWNSTREAM] ###
RG1xx29F063 lora user.notice Jul 9 09:00:09 # PUSH_DATA acknowledged: 100.00%
RG1xx29F063 lora user.notice Jul 9 09:00:09 # PUSH_DATA datagrams sent: 2 (343 bytes)
RG1xx29F063 lora user.notice Jul 9 09:00:09 # RF packets forwarded: 1 (41 bytes)
RG1xx29F063 lora user.notice Jul 9 09:00:09 # CRC_OK: 50.00%, CRC_FAIL: 50.00%, NO_CRC: 0.00%
RG1xx29F063 lora user.notice Jul 9 09:00:09 # RF packets received by concentrator: 2
RG1xx29F063 lora user.notice Jul 9 09:00:09 ### [UPSTREAM] ###
RG1xx29F063 lora user.notice Jul 9 09:00:09 ##### 2021-07-09 08:58:39 GMT #####
RG1xx29F063 lora user.notice Jul 9 09:00:09 INFO: Enabling GPS mode for concentrator's counter.
RG1xx29F063 lora user.notice Jul 9 09:00:09 INFO: host/sx1301 time offset=(1625820936s:351348µs) - drift=19959µs
RG1xx29F063 lora user.notice Jul 9 09:00:09 INFO: Disabling GPS mode for concentrator's counter...
RG1xx29F063 lora user.notice Jul 9 09:00:09 INFO: [down] PULL_ACK received in 26 ms
RG1xx29F063 lora user.notice Jul 9 09:00:09 INFO: Requested 5 power level
RG1xx29F063 lora user.notice Jul 9 09:00:09 INFO: tx_start_delay=1497 (1497.000000) - (1497, bw_delay=0.000000, notch_delay=0.000000)
RG1xx29F063 lora user.notice Jul 9 08:58:21 INFO: Duty = 0, Dwell = 0, Hops = 0, Flags = 0x0, next tx ms = 0
RG1xx29F063 lora user.notice Jul 9 08:58:21 INFO: Channel found, transmission allowed
RG1xx29F063 lora user.notice Jul 9 08:58:21 INFO: Requested time on air: 20
RG1xx29F063 lora user.notice Jul 9 08:58:21 INFO: Requested channel within regulatory rule 902 - 928 MHz
RG1xx29F063 lora user.notice Jul 9 08:58:21 INFO: [down] PULL_ACK received in 28 ms
**RG1xx29F063 lora user.notice Jul 9 08:58:21 JSON down: {"txpk":{"powe":7,"imme":false,"tmst":165190452,"codr":"4/5","datr":"SF8BW500","freq":925.1,"modu":"LORA","rfch":0,"ipol":true,"size":13,"data":"YJvZIwAhAwAG9uREEQ=="}}**
RG1xx29F063 lora user.notice Jul 9 08:58:21 INFO: [down] PULL_RESP received - token[230:33] :)
RG1xx29F063 lora user.notice Jul 9 08:58:21 INFO: [up] PUSH_ACK received in 26 ms
**RG1xx29F063 lora user.notice Jul 9 08:58:21 JSON up: {"rxpk":[{"tmst":163190452,"chan":7,"rfch":0,"freq":904.500000,"stat":1,"modu":"LORA","datr":"SF8BW125","codr":"4/5","lsnr":11.5,"rssi":-75,"size":41,"data":"gJvZIwCAAwADMjqWX6yoV4XJz96iRZOVlo9HJMtXQVxdGeCTRzXoT8c="}]}**
RG1xx29F063 lora user.notice Jul 9 08:58:21 INFO: Received pkt from mote: 0023D99B (fcnt=3)
RG1xx29F063 lora user.notice Jul 9 08:58:21 INFO: [down] PULL_ACK received in 25 ms
RG1xx29F063 lora user.notice Jul 9 08:58:21 INFO: [up] PUSH_ACK received in 27 ms
RG1xx29F063 lora user.notice Jul 9 08:58:21 JSON up: {"stat":{"time":"2021-07-09 08:58:09 GMT","rxnb":2,"rxok":0,"rxfw":0,"ackr":100.0,"dwnb":0,"txnb":0}}

RG1xx29F063 lora user.notice Jul 9 09:07:09 # GPS sync is disabled
RG1xx29F063 lora user.notice Jul 9 09:07:09 ### [GPS] ###
RG1xx29F063 lora user.notice Jul 9 09:07:09 src/jitqueue.c:448:jit_print_queue(): INFO: [jit] queue is empty
RG1xx29F063 lora user.notice Jul 9 09:07:09 # SX1301 time (PPS): 602697652
RG1xx29F063 lora user.notice Jul 9 09:07:09 ### [JIT] ###
RG1xx29F063 lora user.notice Jul 9 09:07:09 # BEACON rejected: 0
RG1xx29F063 lora user.notice Jul 9 09:07:09 # BEACON sent so far: 0
RG1xx29F063 lora user.notice Jul 9 09:07:09 # BEACON queued: 0
RG1xx29F063 lora user.notice Jul 9 09:07:09 # TX rejected (too early): 0.00% (req:4, rej:0)
RG1xx29F063 lora user.notice Jul 9 09:07:09 # TX rejected (too late): 0.00% (req:4, rej:0)
RG1xx29F063 lora user.notice Jul 9 09:07:09 # TX rejected (collision beacon): 0.00% (req:4, rej:0)
RG1xx29F063 lora user.notice Jul 9 09:07:09 # TX rejected (collision packet): 0.00% (req:4, rej:0)
RG1xx29F063 lora user.notice Jul 9 09:07:09 # TX errors: 0
[... cut ...]
format("%s crc %s", RADIO_SET, RADIO_CRC),
format("%s sync %i", RADIO_SET, RADIO_SYNC_WORD),
format("%s wdt %i", RADIO_SET, RADIO_WATCHDOG_TIMEOUT),
"mac set pwridx 5","radio set prlen 8","radio set iqi on",
"sys get hweui","mac set devaddr 0023D99B", "mac set nwkskey 0123456789ABCDEF0123456789ABCDEF" ,
"mac set appskey 0123456789ABCDEF0123456789FEDCBA","mac set dr 4","mac set adr on","mac set retx 5",
"**mac set rxdelay1 2000**","**mac set rx2 8 924500000**","mac set ar on","mac save", "mac join abp"];

mcoracin, 2021-07-13:
@pshingala145 The synchronization between the end-node and the gateway relies on the gateway internal counter precision, which is designed to be accurate enough for LoRaWAN needs.

So in your case, if the end-device never receive the ACK downlinks, you need to ensure that the parameters are aligned between your end-device and the LNS. If it is correct, it could be a RX window opening issue on the end-device side.
Also check that signal is actually visible on the gateway output.

pshingala145, 2021-07-13:
@mcoracin 

When We try the same setup with another gateway node always get ACK as well as Downlink Payload.

Receiving window Parameters are in concordance with the server configuration.

The attached screenshots are the LNS Configuration.   
[image]
[image]

**Lora Receiving Windows Configuration on Device.**
initCommands <- [ format("%s mod %s", RADIO_SET, RADIO_MODE),
format("%s freq %i", RADIO_SET, RADIO_FREQ),
format("%s sf %s", RADIO_SET, RADIO_SPREADING_FACTOR),
format("%s bw %i", RADIO_SET, RADIO_BANDWIDTH),
format("%s cr %s", RADIO_SET, RADIO_CODING_RATE),
format("%s crc %s", RADIO_SET, RADIO_CRC),
format("%s sync %i", RADIO_SET, RADIO_SYNC_WORD),
format("%s wdt %i", RADIO_SET, RADIO_WATCHDOG_TIMEOUT),
"mac set pwridx 5","radio set prlen 8","radio set iqi on",
"sys get hweui","mac set devaddr 0023D99B", "mac set nwkskey 0123456789ABCDEF0123456789ABCDEF" ,
"mac set appskey 0123456789ABCDEF0123456789FEDCBA","mac set dr 4","mac set adr on","mac set retx 5",
"mac set rxdelay1 1000","mac set rx2 8 924500000","mac set ar on","mac save", "mac join abp"];

closed 2022-09-30 by smtc-bot