Lora-net/packet_forwarder #150: JIT error = 5, How to resolve it?

shabaz-git, 2021-12-20:
Hi,

```
Dec 20 17:42:30 raspberrypi sh[1132]:  number of packets 3
Dec 20 17:42:30 raspberrypi sh[1132]: INFO: Received pkt from mote: 00A26052 (fcnt=1764)
Dec 20 17:42:30 raspberrypi sh[1132]: JSON up: {"rxpk":[{"tmst":2617397300,"chan":3,"rfch":1,"freq":865.985000,"stat":1,"modu":"LORA","datr":"SF12BW125","codr":"4/5","lsnr":-16.5,"rssi":-89,"size":23,"data":"gFJgogAA5AYBwIpvAyTPwNHWowIaaDc="},{"tmst":2617397308,"chan":7,"rfch":1,"freq":866.785000,"stat":1,"modu":"LORA","datr":"SF12BW125","codr":"4/5","lsnr":-17.8,"rssi":-91,"size":23,"data":"gFJgogAA5AYBwIpvAyTPwNHWowIaaDc="},{"tmst":2617397308,"chan":5,"rfch":1,"freq":866.385000,"stat":1,"modu":"LORA","datr":"SF12BW125","codr":"4/5","lsnr":7.2,"rssi":-39,"size":23,"data":"gFJgogAA5AYBwIpvAyTPwNHWowIaaDc="}]}
Dec 20 17:42:30 raspberrypi sh[1132]: INFO: [up] PUSH_ACK received in 0 ms
Dec 20 17:42:30 raspberrypi sh[1132]: INFO: [down] PULL_RESP received  - token[19:132] :)
Dec 20 17:42:30 raspberrypi sh[1132]: JSON down: {"txpk":{"imme":false,"rfch":0,"powe":27,"ant":0,"brd":0,"tmst":2618397300,"freq":865.985,"modu":"LORA","datr":"SF12BW125","codr":"4/5","ipol":true,"size":12,"data":"YFJgogCgxwa5x2Y4"}}
Dec 20 17:42:30 raspberrypi sh[1132]: INFO: [down] PULL_RESP received  - token[39:108] :)
Dec 20 17:42:30 raspberrypi sh[1132]: JSON down: {"txpk":{"imme":false,"rfch":0,"powe":27,"ant":0,"brd":0,"tmst":2618397308,"freq":866.385,"modu":"LORA","datr":"SF12BW125","codr":"4/5","ipol":true,"size":12,"data":"YFJgogCgxwa5x2Y4"}}
Dec 20 17:42:30 raspberrypi sh[1132]: src/jitqueue.c:280:jit_enqueue(): ERROR: Packet (type=0) REJECTED, collision with packet already programmed at 2618397300 (2618397308)
Dec 20 17:42:30 raspberrypi sh[1132]: ERROR: Packet REJECTED (jit error=5)
Dec 20 17:42:30 raspberrypi sh[1132]: INFO: [down] PULL_RESP received  - token[39:108] :)
Dec 20 17:42:30 raspberrypi sh[1132]: JSON down: {"txpk":{"imme":false,"rfch":0,"powe":27,"ant":0,"brd":0,"tmst":2619397308,"freq":866.55,"modu":"LORA","datr":"SF8BW125","codr":"4/5","ipol":true,"size":12,"data":"YFJgogCgxwa5x2Y4"}}
Dec 20 17:42:30 raspberrypi sh[1132]: src/jitqueue.c:280:jit_enqueue(): ERROR: Packet (type=0) REJECTED, collision with packet already programmed at 2618397300 (2619397308)
Dec 20 17:42:30 raspberrypi sh[1132]: ERROR: Packet REJECTED (jit error=5)
Dec 20 17:42:30 raspberrypi sh[1132]: INFO: [down] PULL_RESP received  - token[223:16] :)
Dec 20 17:42:30 raspberrypi sh[1132]: JSON down: {"txpk":{"imme":false,"rfch":0,"powe":27,"ant":0,"brd":0,"tmst":2618397308,"freq":866.785,"modu":"LORA","datr":"SF12BW125","codr":"4/5","ipol":true,"size":12,"data":"YFJgogCgxwa5x2Y4"}}
Dec 20 17:42:30 raspberrypi sh[1132]: src/jitqueue.c:280:jit_enqueue(): ERROR: Packet (type=0) REJECTED, collision with packet already programmed at 2618397300 (2618397308)
Dec 20 17:42:30 raspberrypi sh[1132]: ERROR: Packet REJECTED (jit error=5)
Dec 20 17:42:30 raspberrypi sh[1132]: INFO: [down] PULL_RESP received  - token[223:16] :)
Dec 20 17:42:30 raspberrypi sh[1132]: JSON down: {"txpk":{"imme":false,"rfch":0,"powe":27,"ant":0,"brd":0,"tmst":2619397308,"freq":866.55,"modu":"LORA","datr":"SF8BW125","codr":"4/5","ipol":true,"size":12,"data":"YFJgogCgxwa5x2Y4"}}
Dec 20 17:42:30 raspberrypi sh[1132]: src/jitqueue.c:280:jit_enqueue(): ERROR: Packet (type=0) REJECTED, collision with packet already programmed at 2618397300 (2619397308)
Dec 20 17:42:30 raspberrypi sh[1132]: ERROR: Packet REJECTED (jit error=5)

```

The node is configured to retransmit **8 times** if it didn't get the acknowledgement for the **confirmed uplink**.
The received 3 packets are of the same node. Because of this type of behaviour, the node is not getting the ACK. How to solve this issue?

reissjason, 2021-12-22:
You are seeing the same packet from the same device on multiple channels. Only one of the channels is correct as the device can only use one channel to transmit. According to SNR/RSSI this channel was where the device transmitted "freq":866.385000,"lsnr":7.2,"rssi":-39. The other packets are ghosts and should be ignored. The other packets have low RSSI/SNR.

The network server scheduled the downlink for "tmst":2618397300,"freq":865.985, but this is not the channel the device is listening for.

The entire array of "rxpk" needs to be examined for duplicates and choose the best SNR/RSSI to deal with this issue, a simple block for low RSSI/SNR will block legitimate packets for device far from the gateway.

shabaz-git, 2022-01-21:
Hi, 
Thanks for the reply......

But according to the log, it seems the node is not operating on a single frequency (3rd channel starting from zero). Is there any way to make the node work on a single channel? I am operating in the Indian frequency band (IN865).

changes made in **LmHandlerInit()** function
```
        uint16_t channelMask_new[] = { 0x0008};
        mibReq.Type = MIB_CHANNELS_MASK;
        mibReq.Param.ChannelsMask = channelMask_new;
        LoRaMacMibSetRequestConfirm( &mibReq );

        mibReq.Type = MIB_CHANNELS_DEFAULT_MASK;
        mibReq.Param.ChannelsDefaultMask = channelMask_new;
        LoRaMacMibSetRequestConfirm( &mibReq );

```
In  **LoRaMacStatus_t RegionIN865ChannelAdd( ChannelAddParams_t\* channelAdd )** function commented the following line as this is changing the mask.

```
    RegionNvmGroup2->Channels[id].Band = 0;
    //RegionNvmGroup2->ChannelsMask[0] |= ( 1 << id );
    return LORAMAC_STATUS_OK;

```
But the node is unable to join the network and goes into a bad state as there is no default channel for processing the join request. My requirement is after joining the network it should work with the 4th channel. How to achieve that?

closed 2022-09-30 by smtc-bot