Lora-net/packet_forwarder #153: LoRaWAN gateway receives data from another gateway

BlokhinVega, 2022-12-03:
LoRaWAN gateway receiving data from another gateway

Found a problem when gateways accept Downlink messages from other gateways. In the test, 3 gateways are connected to the same network server. One end device is registered on the network server. I send a Link_check from the end device (up_fcnt 7 in the example), and the network server sends a downlink through one of the three gateways (down_fcnt 8 in the example sent by GW#3), which the end device and neighboring gateways successfully receive. This parasitic packet is received by the gateways on a different frequency, but judging by the parameters, this is the same Downlink.

I checked this behavior on multitech mtcdt gateways, mikrotik wAP LoRa 8, Semtech CoreCell Gateway interface board - Everywhere the same behavior. The radio channel reference also does not matter (SX 1301+SX 1257 or SX1302/03 + SX1250) - the problem appears everywhere.

Has anyone experienced similar behavior before?

/84000100 - DevAddr
/
/************************************************************************
/**********************************GW#1********************************
/************************************************************************
/
/INFO: Received pkt from mote: 84000100 (fcnt=7)
/
/JSON up: {"rxpk":[{"tmst":1150502460,"time":"2022-12-02T01:58:25.961506Z","tmms":1353981524961,"chan":4,"rfch":0,"freq":867.300000,"stat":1,"modu":"LORA","datr":"SF12BW125","codr":"4/5","lsnr":5.5,"rssi":-31,"size":14,"data":"QAABAISABwAAShVk0mE="}]}
/INFO: [up] PUSH_ACK received in 2 ms
/INFO: [down] PULL_ACK received in 3 ms
/
/INFO: Received pkt from mote: 84000100 (fcnt=8)
/
/JSON up: {"rxpk":[{"tmst":1152657292,"time":"2022-12-02T01:58:28.116340Z","tmms":1353981527116,"chan":6,"rfch":0,"freq":867.700000,"stat":1,"modu":"LORA","datr":"SF12BW125","codr":"4/5","lsnr":-3.5,"rssi":-99,"size":15,"data":"YAABAISDCAACHwOU/QlG"}]}
/INFO: [up] PUSH_ACK received in 6 ms
/INFO: [down] PULL_ACK received in 3 ms
/
/##### 2022-12-02 01:58:41 GMT #####
/### [UPSTREAM] ###
/# RF packets received by concentrator: 2
/# CRC_OK: 100.00%, CRC_FAIL: 0.00%, NO_CRC: 0.00%
/# RF packets forwarded: 2 (29 bytes)
/# PUSH_DATA datagrams sent: 3 (620 bytes)
/# PUSH_DATA acknowledged: 100.00%
/### [DOWNSTREAM] ###
/# PULL_DATA sent: 2 (100.00% acknowledged)
/# PULL_RESP(onse) datagrams received: 0 (0 bytes)
/# RF packets sent to concentrator: 0 (0 bytes)
/# TX errors: 0
/# TX rejected (collision packet): 0.00% (req:13, rej:0)
/# TX rejected (collision beacon): 0.00% (req:13, rej:0)
/# TX rejected (too late): 0.00% (req:13, rej:0)
/# TX rejected (too early): 0.00% (req:13, rej:0)
/# BEACON queued: 0
/# BEACON sent so far: 0
/# BEACON rejected: 0
/### [JIT] ###
/# SX1301 time (PPS): 1164540943
/src/jitqueue.c:448:jit_print_queue(): INFO: [jit] queue is empty
/### [GPS] ###
/# Valid time reference (age: 0 sec)
/# no valid GPS coordinates available yet
/
/************************************************************************
/**********************************GW#2********************************
/************************************************************************
/
/INFO: Received pkt from mote: 84000100 (fcnt=7)
/
/JSON up: {"rxpk":[{"tmst":605872124,"chan":4,"rfch":0,"freq":867.300000,"stat":1,"modu":"LORA","datr":"SF12BW125","codr":"4/5","lsnr":5.0,"rssi":-42,"size":14,"data":"QAABAISABwAAShVk0mE="}]}
/INFO: [up] PUSH_ACK received in 3 ms
/
/INFO: Received pkt from mote: 84000100 (fcnt=8)
/
/JSON up: {"rxpk":[{"tmst":608026956,"chan":6,"rfch":0,"freq":867.700000,"stat":1,"modu":"LORA","datr":"SF12BW125","codr":"4/5","lsnr":-1.0,"rssi":-69,"size":15,"data":"YAABAISDCAACHwOU/QlG"}]}
/INFO: [up] PUSH_ACK received in 4 ms
/INFO: [down] PULL_ACK received in 2 ms
/
/##### 2022-12-02 01:58:42 GMT #####
/### [UPSTREAM] ###
/# RF packets received by concentrator: 2
/# CRC_OK: 100.00%, CRC_FAIL: 0.00%, NO_CRC: 0.00%
/# RF packets forwarded: 2 (29 bytes)
/# PUSH_DATA datagrams sent: 3 (502 bytes)
/# PUSH_DATA acknowledged: 100.00%
/### [DOWNSTREAM] ###
/# PULL_DATA sent: 3 (100.00% acknowledged)
/# PULL_RESP(onse) datagrams received: 0 (0 bytes)
/# RF packets sent to concentrator: 0 (0 bytes)
/# TX errors: 0
/# TX rejected (collision packet): 0.00% (req:1, rej:0)
/# TX rejected (collision beacon): 0.00% (req:1, rej:0)
/# TX rejected (too late): 0.00% (req:1, rej:0)
/# TX rejected (too early): 0.00% (req:1, rej:0)
/# BEACON queued: 0
/# BEACON sent so far: 0
/# BEACON rejected: 0
/### [JIT] ###
/# SX1301 time (PPS): 602480260
/src/jitqueue.c:448:jit_print_queue(): INFO: [jit] queue is empty
/### [GPS] ###
/# Invalid time reference (age: 1669946322 sec)
/# no valid GPS coordinates available yet
/
/************************************************************************
/**********************************GW#3********************************
/************************************************************************
/
/INFO: Received pkt from mote: 84000100 (fcnt=7)
/
/JSON up: {"rxpk":[{"jver":1,"tmst":109933604,"chan":4,"rfch":0,"freq":867.300000,"mid": 0,"stat":1,"modu":"LORA","datr":"SF12BW125","codr":"4/5","rssis":-54,"lsnr":11.0,"foff":-9443,"rssi":-54,"size":14,"data":"QAABAISABwAAShVk0mE="}]}
/INFO: [up] PUSH_ACK received in 3 ms
/INFO: [down] PULL_RESP received  - token[69:0] :)
/
/JSON down: {"txpk":{"tmst":110933604,"freq":867.3,"rfch":0,"powe":14,"modu":"LORA","datr":"SF12BW125","codr":"4/5","ipol":true,"size":15,"data":"YAABAISDCAACHwOU/QlG","ncrc":false}}
/INFO: [down] PULL_ACK received in 3 ms
/
/##### 2022-12-02 01:58:37 GMT #####
/### [UPSTREAM] ###
/# RF packets received by concentrator: 3
/# CRC_OK: 66.67%, CRC_FAIL: 33.33%, NO_CRC: 0.00%
/# RF packets forwarded: 2 (26 bytes)
/# PUSH_DATA datagrams sent: 3 (595 bytes)
/# PUSH_DATA acknowledged: 100.00%
/### [DOWNSTREAM] ###
/# PULL_DATA sent: 3 (100.00% acknowledged)
/# PULL_RESP(onse) datagrams received: 1 (174 bytes)
/# RF packets sent to concentrator: 1 (15 bytes)
/# TX errors: 0
/# TX rejected (collision packet): 0.00% (req:4, rej:0)
/# TX rejected (collision beacon): 0.00% (req:4, rej:0)
/# TX rejected (too late): 0.00% (req:4, rej:0)
/# TX rejected (too early): 0.00% (req:4, rej:0)
/### SX1302 Status ###
/# SX1302 counter (INST): 120826979
/# SX1302 counter (PPS):  0
/# BEACON queued: 0
/# BEACON sent so far: 0
/# BEACON rejected: 0
/### [JIT] ###
/src/jitqueue.c:440:jit_print_queue(): INFO: [jit] queue is empty
/#--------
/src/jitqueue.c:440:jit_print_queue(): INFO: [jit] queue is empty
/### [GPS] ###
/# Invalid time reference (age: 1669946317 sec)
/# no valid GPS coordinates available yet
/
/************************************************************************
/************************************************************************
/************************************************************************

reissjason, 2022-12-03:
A LoRa downlink packet should have CRC off. So set "ncrc": true, the field is no-crc so true disables it. This will cause the other gateways to reject the packets as they expect CRC enabled for uplinks. I assume the gateways are close to each other, did all SF settings show this or only SF12? 

See page 89 of RP2 1.0.4
https://resources.lora-alliance.org/technical-specifications/rp002-1-0-4-regional-parameters

BlokhinVega, 2022-12-06:
> A LoRa downlink packet should have CRC off. So set "ncrc": true, the field is no-crc so true disables it. This will cause the other gateways to reject the packets as they expect CRC enabled for [...]

Thanks for your reply.

Yes, in the test, the gateways are installed in the same building. But this situation also occurs in real conditions. This gives an increase in traffic to the network server.

Checked on all SF. The problem is reproduced everywhere.

You're right! I set "ncrc": true and now I don't receive such packets on the network server. Works great!