Lora-net/packet_forwarder #157: LoRaWAN Class B - No Beacons Received

Mariiian09, 2025-04-02:
Hello everyone,

I am working with an end device using the I-CUBE-LRWAN Expansion Package (LoRaWAN v1.0.3) and configuring it for Class B operation. However, my device is not receiving any beacons.

After investigating, I found old issues stating that the gateway implementation is designed for LoRaWAN 1.1, and some modifications are required to make it work with LoRaWAN 1.0:

    "You may need to slightly adapt it if used with LoRaWAN1.0."

I would appreciate any guidance on what changes are needed to make the gateway and end device work properly with LoRaWAN 1.0.3 in Class B mode. Has anyone faced a similar issue or successfully adapted the setup?

Thanks in advance!

mcoracin, 2025-04-02:
Hello,
Which LoRaWAN region are you using for your tests ?
The packet forwarder beacon should work with LoRaWAN 1.0.3, it is used by our internal validation team to check Class B with our LoRaWAN stack (LoRa Basics Modem).
Best regards,
Michael

Mariiian09, 2025-04-03:
> Hello, Which LoRaWAN region are you using for your tests ? The packet forwarder beacon should work with LoRaWAN 1.0.3, it is used by our internal validation team to check Class B with our LoRaWAN stack (LoRa Basics Modem). Best regards, Michael

Hello Michael,

Thank you for your response.

I am using the EU868 region for my tests. My device is configured for LoRaWAN 1.0.3 and Class B, with the following settings:

    Ping-slot periodicity: 8

    Ping-slot data-rate: 3

    Ping-slot frequency: 869525000 Hz

I can see the beacon being transmitted by the gateway:
```

Apr 03 10:13:51 Lorank8 lorank8v1[1373]: ##### 2025-04-03 10:13:27 GMT #####
Apr 03 10:13:51 Lorank8 lorank8v1[1373]: ### [UPSTREAM] ###
Apr 03 10:13:51 Lorank8 lorank8v1[1373]: # RF packets received by concentrator: 3
Apr 03 10:13:51 Lorank8 lorank8v1[1373]: # CRC_OK: 33.33%, CRC_FAIL: 66.67%, NO_CRC: 0.00%
Apr 03 10:13:51 Lorank8 lorank8v1[1373]: # RF packets forwarded: 1 (29 bytes)
Apr 03 10:13:51 Lorank8 lorank8v1[1373]: # PUSH_DATA datagrams sent: 2 (425 bytes)
Apr 03 10:13:51 Lorank8 lorank8v1[1373]: # PUSH_DATA acknowledged: 100.00%
Apr 03 10:13:51 Lorank8 lorank8v1[1373]: ### [DOWNSTREAM] ###
Apr 03 10:13:51 Lorank8 lorank8v1[1373]: # PULL_DATA sent: 3 (100.00% acknowledged)
Apr 03 10:13:51 Lorank8 lorank8v1[1373]: # PULL_RESP(onse) datagrams received: 1 (193 bytes)
Apr 03 10:13:51 Lorank8 lorank8v1[1373]: # RF packets sent to concentrator: 1 (18 bytes)
Apr 03 10:13:51 Lorank8 lorank8v1[1373]: # TX errors: 0
Apr 03 10:13:51 Lorank8 lorank8v1[1373]: # TX rejected (collision packet): 0.00% (req:11, rej:0)
Apr 03 10:13:51 Lorank8 lorank8v1[1373]: # TX rejected (collision beacon): 0.00% (req:11, rej:0)
Apr 03 10:13:51 Lorank8 lorank8v1[1373]: # TX rejected (too late): 0.00% (req:11, rej:0)
Apr 03 10:13:51 Lorank8 lorank8v1[1373]: # TX rejected (too early): 0.00% (req:11, rej:0)
Apr 03 10:13:51 Lorank8 lorank8v1[1373]: # BEACON queued: 8
Apr 03 10:13:51 Lorank8 lorank8v1[1373]: # **BEACON sent so far: 5**
Apr 03 10:13:51 Lorank8 lorank8v1[1373]: # BEACON rejected: 0
Apr 03 10:13:51 Lorank8 lorank8v1[1373]: ### [JIT] ###
Apr 03 10:13:51 Lorank8 lorank8v1[1373]: # SX1301 time (PPS): 782426248
Apr 03 10:13:51 Lorank8 lorank8v1[1373]: src/jitqueue.c:461:jit_print_queue(): INFO: [jit] queue contains 3 packets:
Apr 03 10:13:51 Lorank8 lorank8v1[1373]: src/jitqueue.c:462:jit_print_queue(): INFO: [jit] queue contains 3 beacons:
Apr 03 10:13:51 Lorank8 lorank8v1[1373]: src/jitqueue.c:465:jit_print_queue():  - node[0]: count_us=815431934 - type=3
Apr 03 10:13:51 Lorank8 lorank8v1[1373]: src/jitqueue.c:465:jit_print_queue():  - node[1]: count_us=916430685 - type=3
Apr 03 10:13:51 Lorank8 lorank8v1[1373]: src/jitqueue.c:465:jit_print_queue():  - node[2]: count_us=1044429376 - type=3
Apr 03 10:13:51 Lorank8 lorank8v1[1373]: ### [GPS] ###
Apr 03 10:13:51 Lorank8 lorank8v1[1373]: # Valid time reference (age: 0 sec)
Apr 03 10:13:51 Lorank8 lorank8v1[1373]: # GPS coordinates: latitude 38.97756, longitude -1.85721, altitude 739 m
Apr 03 10:13:51 Lorank8 lorank8v1[1373]: ##### END #####
Apr 03 10:13:51 Lorank8 lorank8v1[1373]: JSON up: {"stat":{"time":"2025-04-03 10:13:27 GMT","lati":38.97756,"long":-1.85721,"alti":739,"rxnb":3,"rxok":1,"rxfw":1,"ackr":100.0,"dwnb":1,"txnb":1}}
Apr 03 10:13:51 Lorank8 lorank8v1[1373]: INFO: [up] PUSH_ACK received in 1 ms
Apr 03 10:13:51 Lorank8 lorank8v1[1373]: INFO: [down] PULL_ACK received in 0 ms
Apr 03 10:13:51 Lorank8 lorank8v1[1373]: INFO: [down] PULL_ACK received in 0 ms
Apr 03 10:13:51 Lorank8 lorank8v1[1373]: INFO: Received pkt from mote: 0111A5B4 (fcnt=26)

```
But my end device is not receiving it. Is there any specific configuration or parameter that I should check to ensure the device can synchronize correctly with the beacon?
Best regards,
Marian

Mariiian09, 2025-04-04:
> Hello Marian, Ok, so I don't think that it is an issue with the beacon format emitted by the gateway. The gateway side is quite simple for Class-B, as long as you have a robust GPS reception. I would rather suspect an issue on the end-device side, as it is generally more tricky for the synchronization. Have you wait long enough for the sync to happen ? I think that I-CUBE-LRWAN Expansion Package is based on the LoRaMac-node stack but it has been integrated by STM, I don't know what is the status exactly. :/ You may need to contact STM support for this. May I ask what is the application you need class-b for ? Best regards, Michael

Hello Michael,

Thank you for your suggestions. I already contacted STM support through their forum, and some users mentioned that the issue might actually be on the gateway side.

After further investigation, I realized that the GPS PPS (Pulse Per Second) pin is not being used in my current setup. From what I've read, this signal is critical for Class B synchronization, as it ensures precise timing for beacon transmission. Could this be the root cause of my device not receiving the beacons?

Currently, the PPS signal is derived from the chip itself, but I believe it would be more accurate to use the PPS provided by the GPS module—though I’m not sure how to configure that.

It’s worth noting that my application is solely focused on researching the LoRaWAN Class B standard. The goal is not to deploy a production system, but rather to study how the synchronization mechanisms work in practice.

I’d greatly appreciate any guidance you can offer.

Best regards,
Marian

mcoracin, 2025-04-04:
Hello Marian,
Oh, for sure, for class-B, it is very important to have the PPS pin from the GPS receiver connected to the sx1301 chip of the gateway. The beacon is send on a PPS pulse by the sx1301, so if there is no PPS, the beacons won't be sent.
Do you have a spectrum viewer (a rtl-sdr dongle, a hackrf....) to check what is actually being emitted in the air ?
Which hardware are you using exactly for the gateway ?
Best regards,
Michael

Mariiian09, 2025-04-07:
> Hello Marian, Oh, for sure, for class-B, it is very important to have the PPS pin from the GPS receiver connected to the sx1301 chip of the gateway. The beacon is send on a PPS pulse by the sx1301, [...]

Hello Michael,

Thanks again for your help.

Regarding your suggestion, I'm trying to ensure the GPS synchronization is as accurate as possible. I have connected my GPS module to the BeagleBone Green as shown in the table below. However, I'm not sure which pin I should use to connect the PPS signal from the GPS to the BeagleBone so that the packet forwarder can properly use it for beacon scheduling.

Here are my current physical connections:

[image]

I don't have a spectrum analyzer, but I do have a logic analyzer. I'm currently monitoring the beacon transmission using a GPIO pin that I manually set high when the beacon is queued, and comparing that against the reception windows of the end device. However, the end device's window never aligns with the supposed beacon.

Do you know which BBG pin I should connect the PPS signal to? Or how to properly configure the gateway to use it?

Thanks in advance!

Best regards,
Marian

mcoracin, 2025-04-07:
Hello Marian,
The important pin to be connected is the PPS pin from the GPS module to the iC880A, so that the PPS is directory connected to the sx1301.

You can refer to the iC880A datasheet here:
https://wireless-solutions.de/data/Wireless%20Solutions/Downloads/iC880A/iC880A-SPI-Documents.zip
According to this document, the PPS pin of your GPS receiver should be connected to the pin 19 of the iC880A.

It is also described here (not specific to basicstation, just check the hardware connexion part, see the "Class B support" section):
https://lora.vsb.cz/index.php/868-mhz-ic880a-lora-basics-station/

Best regards,
Michael

Mariiian09, 2025-04-07:
> Hello Marian, The important pin to be connected is the PPS pin from the GPS module to the iC880A, so that the PPS is directory connected to the sx1301.
>
> You can refer to the iC880A datasheet here: [...]

Hello Michael,

I wanted to sincerely thank you for your valuable help in solving the issue with my setup. After following your guidance, I was able to resolve the problem, and the device is now successfully receiving the beacons.

Your suggestions about the PPS signal and its importance for synchronization were crucial, and I really appreciate your patience and support throughout this process.

Thank you once again!

Best regards,
Marian

mcoracin, 2025-04-07:
Hello Marian,
Great news ! Enjoy experimenting with Class-B ;)
Best regards,
Michael